craftbeerpi/craftbeerpi #5: Umstellung auf dev2.1

henneonrails, 2016-04-16:
`pi@raspberrypi:~/craftbeerpi $ sudo python runserver.py 
Traceback (most recent call last):
  File "runserver.py", line 4, in <module>
    from brewapp import app, socketio
  File "/home/pi/craftbeerpi/brewapp/__init__.py", line 50, in <module>
    from .base.views import base
  File "/home/pi/craftbeerpi/brewapp/base/__init__.py", line 6, in <module>
    import kettle
  File "/home/pi/craftbeerpi/brewapp/base/kettle.py", line 9, in <module>
    from brewapp.base.automatic.automaticlogic import *
  File "/home/pi/craftbeerpi/brewapp/base/automatic/__init__.py", line 4, in <module>
    import custom
ImportError: No module named custom
pi@raspberrypi:~/craftbeerpi $ git status
Auf Branch dev2.1
Ihr Branch ist auf dem selben Stand wie 'origin/dev2.1'.
nichts zu committen, Arbeitsverzeichnis unverändert
pi@raspberrypi:~/craftbeerpi $ 
`

Manuel83, 2016-04-16:
Ist gefixt. "custom" ist entfernt. :-) Sorry.

henneonrails, 2016-04-16:
```
Traceback (most recent call last):
  File "runserver.py", line 4, in <module>
    from brewapp import app, socketio
  File "/home/pi/craftbeerpi/brewapp/__init__.py", line 90, in <module>
    i.get("function")()
  File "/home/pi/craftbeerpi/brewapp/base/config.py", line 42, in init
    readConfig()
  File "/home/pi/craftbeerpi/brewapp/base/config.py", line 27, in readConfig
    config = Config.query.all()
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2398, in all
    return list(self)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2515, in __iter__
    return self._execute_and_instances(context)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2530, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 914, in execute
    return meth(self, multiparams, params)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/sql/elements.py", line 323, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1010, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1146, in _execute_context
    context)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1339, in _handle_dbapi_exception
    exc_info
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/util/compat.py", line 199, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
    context)
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/default.py", line 442, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: config.type [SQL: u'SELECT config.name AS config_name, config.type AS config_type, config.value AS config_value, config.description AS config_description \nFROM config']

Manuel83, 2016-04-16:
Es hat größere Änderungen am Datenbank Schema gegeben. Bitte die craftbeerpi.db löschen und neu starten.

closed 2016-04-16 by henneonrails

MrLeeh, 2016-04-18:
Für den Fall, dass Du auch in Zukunft Änderungen an der Datenbank planst, kann ich nur die Flask-Extension [Flask-Migrate](https://flask-migrate.readthedocs.org/en/latest/) empfehlen. Die erstellt automatisch Update-Scripts aus deinem aktuellen SQLAlchemy-Modell und hält die Datenbank konsistent zum Modell.

Manuel83, 2016-04-18:
Cool danke für den Hinweis.

MrLeeh, 2016-04-18:
Gern, falls Du Hilfe brauchst sag Bescheid. Ich nutze das Modul auch für meinen Blog.