craftbeerpi/craftbeerpi #6: Uhrzeit

henneonrails, 2016-04-18:
Es wäre schön, wenn bei den Zeiten die Systemzeit genommen würde. Im Moment sieht es mir so aus als würde hier die GMT benutzt.

Manuel83, 2016-04-18:
Problem ist nachvollziehbar. Ich arbeite an einer Lösung

Manuel83, 2016-04-18:
Intern werden die Zeiten in UTC gespeichert daher kommt das Problem.

Manuel83, 2016-04-22:
Sollte jetzt im branch 2.1 gefixt sein. Bitte um Feedback

henneonrails, 2016-04-22:
[image]

ich habe mal ein Maischeprogramm per Hand durchgeklickt. Es sieht so aus als ob es noch nicht funktioniert. Alle Zeiten sind bei mir noch UTC obwohl der PI auf CEST eingestellt ist.

  ```
 pi@raspberrypi:~ $ date
 Fr 22. Apr 23:57:46 CEST 2016
```

henneonrails, 2016-04-22:
oder muss ich die Datenbank wieder löschen?

Manuel83, 2016-04-23:
Die Datenbank muss man nicht löschen. Ich schau noch mal drauf.

chixxi, 2016-04-24:
Scheint in commit a9fbf99 auch noch fehlerhaft zu sein, 2 Stunden Offset.

Gruss chixxi

Manuel83, 2016-04-25:
Das war echt doof. 
1. Python führt erstmal keine Timezone.
2. Beim speichern der Zeiten in der Datenbank wurde eine extra gesetzte Zeitzone abgeschnitten.

Bevor die Daten jetzt an den Browser übertragen werden, wird UTC als Zeitzone einfach fest angehängt. 
Im Browser wird dann auf die lokale Zeit umgerechnet. 

Bitte um Feedback ob es jetzt funktioniert.

CK-AKS, 2016-04-26:
Soeben habe ich mal händisch ein Rezept mit angeschlossenem Temperaturfühler durchgespielt. Bei der ersten Rast (Einmaischen) wurde wiederum die Zeit um 2h zurückgesetzt, die folgenden Rasten haben sich dann wohl auf diese Anfangsstartzeit bezogen (sh. Screenshot: Bei der Gummirast im Rezept nimmt das Programm nicht die 45min, sondern 2:45h).
[image]

Manuel83, 2016-04-28:
Fixed

closed 2016-04-28 by Manuel83