craftbeerpi/craftbeerpi #15: Can not delete kettle when name is empty

Manuel83, 2016-04-28:
Can not delete kettle when name is empty

DaSchaef, 2016-05-04:
I can't reproduce this? (would be willing too help)

danopat, 2016-05-04:
Same problem here using craftbeerpi 2.1. In GUI you have to click to the kettle name to delete it, when you create a kettle with no name, there is no text to click on so you can not edit/delete it.

DaSchaef, 2016-05-05:
Hmm, I'm on 2.2 and I'm too stupid to create a kettle without name. The problem is that I can't create it, if the name is empty, ne kettle is not created - Manuel, did you already fix it?

Manuel83, 2016-05-06:
This bug was fixed during refactoring of kettle dialogs for v2.1. 
In the past the "edit kettle dialog" was a separate screen. This separate screen had the bug. I think we can close this issue.

closed 2016-05-06 by Manuel83