craftbeerpi/craftbeerpi #20: name thermometers

franktegtmeyer, 2016-05-02:
add a function to rename thermometers in the "hardware" tab

DaSchaef, 2016-05-03:
I'm not totally sure, but I think the name of the thermometer seems to be generated in the software/hw-driver layer (meaning e.g. the OneWire Bus Address).
What is the use case for a custom name? 
E.g. If I rename a thermometer to "Upper Thermoelement" in the "hardware" tab, then I still need to know what OneWire Bus Address it relates to (for correct HW connection)?

Manuel83, 2016-05-06:
I don't understand the use case.

DaSchaef, 2016-05-16:
No feedback :-(
I would vote for closing.

franktegtmeyer, 2016-05-16:
When using more than one probe, I'd like to put a label on the cable and type in the name in the app. Would make it much more convienient than with the cryptic bus ids.... 😉 
Sorry... I'm on holiday right now.

Manuel83, 2016-07-17:
Added to version 2.2

closed 2016-07-17 by Manuel83