craftbeerpi/craftbeerpi #29: Heating

o5i, 2016-05-11:
Hi... I have just a question about the heating...
Is the heating power regulated trough PWM?
I habe a BIAB setup where i need to do that...
thanks

Manuel83, 2016-05-11:
Hi
currently CraftBeerPI is supporting a PID and Overshoot Logic out of the box. Can you tell me some more details about your hardware? 
In generell its very easy to add further heating logics to CBP. With some basic pyhton skills a custom PWM based logic is not a big deal.

Here you will find the skeleton for a custom logic.

https://github.com/Manuel83/craftbeerpi/blob/master/brewapp/base/automatic/examplelogic.py

Here you will find an example how programm PWM with in Python
https://sourceforge.net/p/raspberry-gpio-python/wiki/PWM/

I'm currently implementing the PWM support for the agitator.

awillems, 2016-08-20:
@o5i  Yo I had the same issue I'm using a DIY rims tube with a 2000 watt, 120 volt element. and then another 2000 watt, 120 volt element for my hot liqueur tank for sparging and SSR's to control them. Here is what I did. Its hard coded so its not ui adjustable, and you may want to adjust the frequency and operational cycles to what suits your needs. 
[gpio.py.txt](https://github.com/Manuel83/craftbeerpi/files/428068/gpio.py.txt)
[image]

taloever, 2016-09-14:
Hi,
To have the possibillity to do the BOIL with PWM would be great. Doing my build these days, almot ready to test a 3 ketle+ setup.
 TomA

piitsi, 2016-10-28:
Hi, yes some kind of heating PWM or cycle or something woul be nice. I have 2.8kw boiling kettle and boil is just too high.