craftbeerpi/craftbeerpi #44: Sub-Steps

tomandre76, 2016-05-25:
As someone mentioned on Facebook. Sub-steps. I would like to have the ability to initiate heating of the spargewater in HLT at the same time as I do the Mash in another kettle.

bjornarfonn, 2016-05-26:
There are some difficulties in running a HERMS setup, or at least from my testing. Because the steps only allows for one kettle pr step. This means, if i want to activate the HLT kettle during the mashing, i have to set the mash kettle`s Heater GPIO to the same as the HLT. That will initiate the HLT kettle during mashing. Could be resolved by pre-defining the HLT, Mashtun and boiler kettle during the initial setup. This way, the mash tun`s temperature will be controlled by the HLT, not by itself. :)

alanrflisboa, 2016-05-28:
HI.I suggest, in Facebook, that sub-steps are included in the process. For example, a heater to be activated at some time in the process of another kettle. As in the picture:

[image]

owold, 2016-07-18:
I think sub steps could allow for automatic sparging as well. In my setup, opening two valves and starting two pumps will start autosparge. For people with gravity fed setup this would be even more trivial. 

But this will require steps that just flip/set a switch/gpio, and then continue execution.

Twit3986, 2017-05-09:
Has there been progress with this? Could even just be done by linking the start of a step to another step so they run in parallel with the option of having a delayed start to the step.