craftbeerpi/craftbeerpi #45: Control Box LCD Display

blazem, 2016-05-28:
As per my facebook post, can you guys add the ability to show short brew process description with counter and current sequence step, next step etc on an LCD screen. Ideally 4 lines?

Manuel83, 2016-05-28:
This is already implementiert by the German community. I will add it to the dev2.2 branch soon.

http://hobbybrauer.de/forum/download/file.php?id=17764

blazem, 2016-05-29:
Awesome, thank you for this.

paalja, 2016-06-17:
do you know if there will be support for IC2?
the linked code didn't seem to support it. Just to save IO pins on the board :)

rampanelli, 2017-03-13:
Is there any electrical scheme for using a 20x4 LCD display to craftbeerpi?

blazem, 2017-03-13:
You use the IC2 pins on the raspi board.

rampanelli, 2017-03-13:
Simultaneous to video display? Not config needed?

blazem, 2017-03-13:
you will need to make sure the i2c is activated on the pi.

rampanelli, 2017-03-13:
I'll try, thank you.

rampanelli, 2017-03-14:
Thanks, but how do I enable ic2 on crafbeerpi?

cow77, 2017-03-14:
raspi-config: enable i2c