craftbeerpi/craftbeerpi #48: dev 2.2 - just background-image

tomandre76, 2016-06-02:
Just getting the background-image in Safari on my Mac. Chrome and Firefox works fine. On my iPhone I also just get det background-image in Safari and also Chrome. This did work fine in 2.1

henneonrails, 2016-06-04:
+1

Safari on iPhone only showing the background image. No dashboard or other interface.

[image]

Manuel83, 2016-06-04:
Fixed this morning. Please do a `git pull `on dev2.2

closed 2016-06-04 by Manuel83

tomandre76, 2016-06-04:
How to do a git pull ?

Manuel83, 2016-06-04:
Navigate into the CraftBeerPi installation folder and call "git pull"

tomandre76, 2016-06-06:
Sorry Manuel, i´m not so into this Raspbian. Do you mean navigate in to the folder with Terminal on the Raspberry PI ? Or ? Can you please give me a more detailed explanation. From the noob :)

Manuel83, 2016-06-06:
Open a terminal window on the RPi and navigate into the craftbeerpi installation folder. Default is /home/pi/craftbeerpi
Inside this folder call `git pull`

tomandre76, 2016-06-06:
So I just write: git pull   in that folder ?

tomandre76, 2016-06-06:
Had to write sudo git pull :)

Manuel83, 2016-06-06:
Oh sorry :-)

anthonyh1964, 2016-07-03:
Hi there, how do I install 2.2 version, I've tried downloading the zip file and scp to the craftbeerpi folder.

Cheers

Tony

bjornarfonn, 2016-07-03:
Tony, navigate to your craftbeerpi folder and do: 
git checkout <branch> 
So for dev2.2 then do: git checkout dev2.2

anthonyh1964, 2016-07-03:
Hi
So, I'm SSH into folder /home/pi/craftbeerpi and once in the folder I run the command sudo git checkout dev2.2

I get a "not a git repository" message

Cheers

anthonyh1964, 2016-07-03:
OK I seem to be getting somewhere, but, the dashboard is blank

anthonyh1964, 2016-07-03:
OK, the files seem to have updated, but, cannot run the session, the service does not appear to start, I've manually run sudo python runserver.py to start the service but, cannot access

bjornarfonn, 2016-07-03:
Make sure when you are accessing the dashboard with a browser, that you are using "private mode" or "incognito" so that its not trying to display cached content. If you have tried to reboot the raspberry pi and still cant access it, you can easily run the installer again from the folder

bjornarfonn, 2016-07-03:
Also, if the dashboard is not showing, try to run the setup again http://ip:5000/base/setup

ivanbate, 2016-07-05:
hi there,
 i had the same issue when running "sudo git checkout dev2.2". It  updated
but I could not load the 2.2 app via web browser. I manually started the
server and that fails with an error msg. Screenshot attached with the
error. Something about the SwaggerAPI module!!

I just reinstalled 2.1 and that is now working again!!

Manuel83, 2016-07-07:
I will have a look on this.
Sorry for this inconvenience