craftbeerpi/craftbeerpi #51: Unable to configure kettle agitator / heater with PiFace

chixxi, 2016-06-04:
Wollte eigentlich Morgen mit der Version dev2.2 commit f61974e brauen und das wieder mal richtig testen. Leider ist es nicht möglich die PiFace Ausgänge in dem Setup der Kettles zu machen, es wird einfach keine Auswahl angeboten:

[image]

Interessanterweise klappt das aber einwandfrei bei der Ansteuerung der zusätzlichen Hardware:
[image]

Habe verschiedene Browser getestet und cache, cookies usw. gelöscht.

Manuel83, 2016-06-04:
Hi 
in dev2.2 the concept changed a little bit. 
You have to configure your heater and agitator first in the hardware tab. After this you can select the configured hardware at the kettle setup.

Currently this may look complicated but it will change during further during the dev2.2 development.

Manuel83, 2016-06-04:
Let me know if its working

chixxi, 2016-06-04:
Works just fine. Thank you for the fast response. I will send further feedback after brewing tomorrow.

closed 2016-06-04 by chixxi