craftbeerpi/craftbeerpi #52: Support for multi-step or step-less controllable heaters

FredWanninger, 2016-06-04:
We use an induction cooker in our brew setup. We made the induction cooker controllable via relays that "simulate" pressed buttons on the button panel of the cooker. We can now control the cooker via a custom control software in 9 steps.
We hacked your software to support our setup. In particular, dummy GPIO and PID logic write the heater state (output power percentage) to a file, our control software reads the file, and translates the state to a key press combination for the induction cooker.
The solution works, but we would prefer a more sophisticated implementation (The heater button in the webinterface could for example show the current heater output power).
Did you foresee support for multi-step or step-less controllable heaters in your design? I'm afraid the architecture is quite bound to using exactly one GPIO for a heater that can only be switched on or off. Maybe you see an easy and elgant way to add support for our setup in your software?

Manuel83, 2016-06-05:
Hi,
I like to understand your current setup in more detail.
Can you provide me your modified dummy GPIO and PID Logic file? I 
Please send the files to manuel@craftbeerp.com

Regards,
Manuel

FredWanninger, 2017-06-26:
Support for multi-step or step-less controllable heaters now possible in Craftbeerpi3 :+1:

closed 2017-06-26 by FredWanninger