craftbeerpi/craftbeerpi #56: Add password security to web login home page

ivanbate, 2016-06-14:
Would be good if there was a password / username entry screen that can be added so when you open the app to the internet there is a basic security challenge.

Manuel83, 2016-06-16:
Good Idea will be added during dev2.2.

DrLaban, 2016-06-21:
Maybe you also should add the possibility to use HTTPS - if people want to make it available through the Internet without sending username/pwd in clear text

DaSchaef, 2016-06-29:
You could easily add a nginx proxy to give the internet connection https and authentication. I personally would focus to other issues first :)

e.g.
https://www.digitalocean.com/community/tutorials/how-to-configure-nginx-with-ssl-as-a-reverse-proxy-for-jenkins

ivanbate, 2016-08-24:
Hi Manuel
I can see in Dev2.2 that you can add a Username / Password in the Configuration section. What i cant work out is how to enable the requirement to enter username password. It would be good if you you could choose to enable the requirement for authentication or not enable it. Im sure some CBPI users don't want the authentication (for exmplae when running over local home wifii) but others who run it over the internet probably do want it.
Thanks,
Ivan

Manuel83, 2016-08-24:
Currently the function is commented out. I'm not happy with the basic authentication. I like to have a nice login dialog.

ivanbate, 2017-05-30:
Hi Manuel
I am unable to get the authentication working. Not sure if i understand what i am supposed to do. I can see the settings in the configuration section but don't get asked to authenticate either over local WiFi or over the internet. I notice some other user did get prompted.

Am i doing something wrong or is this feature still commented out as per your previous comment.

I would be happy to add instructions to your wiki if i knew how to use the authentication feature.

Thanks for the app. I love it.
Ivan

ivanbate, 2018-03-23:
Hi Manuel
Any update about the password security. I see the setting in the Configuration page but they dont seem to do anything
thanks
ivan