craftbeerpi/craftbeerpi #60: Add support for custom hardware as extra buttons

DrLaban, 2016-06-21:
Is it possible to add a feature which makes it possible to add custom hardware as buttons to the kettles?

- Extra heater
- Pump
- Valve
- Etc

Also with the possibility to choose the icon

Manuel83, 2016-06-21:
Currently you can add a heater or agitator to a kettle. Custom hardware like pumps or valves can be add just separately. You can control them via dashboard. If I understand your request correct you just like to have them display in a kettle panel right?

DrLaban, 2016-06-22:
Yepp. Thats right.
And if you assign it to a kettle panel it will not show up under the "additional hardware" panel".

Manuel83, 2016-06-22:
In the next version you can select if a hardware should be displayed or hide in the dashboard. 

So you have the option to add it to a kettle and in addition you can display it under hardware

Blom1974, 2016-08-22:
This functions seems to work great in dev2.2 branch