craftbeerpi/craftbeerpi #63: Database error after changing to dev2.2

bjornarfonn, 2016-07-19:
Trying to switch to the new dev2.2 and pulling the new files, the server wont start. 
during a manual "runserver.py" it ends up with an sqlite3 error;

sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: hardware.config [SQL: u'SELECT hardware.id AS hardware_id, hardware.name AS hardware_name, hardware.type AS hardware_type, hardware.config AS hardware_config \nFROM hardware'] 

It seems to me as this would be corrected if we installed the software in 2.2 version, as the tables and columns would be created as the initial setup, but not during a switch from master to dev?
There are several changes on the folder structure as well, so might be a simple fix to where its refering to an old path?

Manuel83, 2016-07-19:
The table structure in the lastest dev 2.2 changed. For the current beta I recommand to delete the craftbeerpi.db and restart the system. Later we need a migration script

bjornarfonn, 2016-07-19:
That worked to get it running. Alltough, it doesnt allow to get pass the "lets go brewing" section. The button doesnt initiate anything. Tried in all languages, with the different configurations, and it still doesnt allow me to progress. Stopped the craftbeerpiboot service and started with manual command in -verbose. When the "Lets go brewing" is pressed, it outputs;
 " 'wsgi.version': (1, 0)} failed with KeyError"

tomandre76, 2016-07-31:
I have the same trouble with "lets go brewing". Any solutions ?

closed 2016-08-14 by Manuel83