craftbeerpi/craftbeerpi #67: Licensing of craftbeerpi forbids any changes

stone, 2016-08-03:
Creating an Issue to follow up on the discussions on the facebook-page. The current license is very restrictive and makes it impossible to change the code and restricts its usage. In the spirit of many other projects like this one (brewpi, brewfactory..) please consider changing the license to an Open Source / Free Software one to encourage developers and other to join in and contribute to the project.

guysoft, 2017-10-12:
Hey, same issue here.
1. I want to fork and add support for SC2262 RF relays, but the license states ``It's not permitted to distribute the modified software. Modification can be distributed via the official CraftBeerPi release only.``.
This license makes no sense, it means that all 116 forks to date are illegal. 
2. I wrote a raspberrypi distro that ships with craftbeerpi pre-installed, similar to [OctoPi](https://octopi.octoprint.org/), but that can't be done neither.

If you want to stop non-commercial and want to make sure you are attributed correctly, I'd recommend using something such as Creative Commons. Moreover, If someone actually truly violates this you will actually have organizations that could help you legally.

@Manuel83 Can you please comment on this?

archonic, 2017-10-12:
I'm a full-time software engineer and I'd love to contribute to this repo and make it the best it can be, but my ability and motivation is severy hampered by the license. Limiting peoples contributions to plugins will really hurt quality over time.

Manuel83, 2017-10-12:
Well, I would like to change the license but I have one issue to be complete open source and free. 
CraftBeerPi is designed for home brewers. Unfortunately it turned out that commercial users tried to package CraftBeePi and sell it standalone with preinstalled SD cards or build entire brewing controllers for a price of $500 or more without supporting the CBPi Project in any way. This is not the intention of CraftBeerPI and from my point of view not fair. I take my private money to buy all that hardware required to build this solution. I'm open to any license change. For example where private home brewers can change the code and install as they like but I want to avoid that other making profit out of a community project. I'm a simple software developer and this is my first community project. 

I experienced once a project in the java environment. As small community from Turkey build an excellent UI framework. Unfortunately published with wrong license. There was one bigger commercial company. They just branched the code,  rebranded it and sold it with a commercial license.

guysoft, 2017-10-12:
Ok, this is really good news. Because it means we can solve this and reach an understanding.

You could make it non-commercial share-alike for a start. It also means you need to give attribution:
https://creativecommons.org/licenses/by-nc-sa/3.0/de/deed.en

@foosel , the creator of OctoPrint uses AGPL. ir requires anyone that makes a change to release the code. 
She might have advice because she had OctoPrint sponsored by BQ. If I catch her.

Anyway sounds like what you want is CC 3.0 share alike non-commercial. What do you think?

stone, 2017-10-13:
But there is no problem to use others work and build upon that? 

This project is standing on the shoulders of many different Open Source / Free Software projects which have given you the right to use their software as you wish. They day when you loose interest in the project it will impossible to keep it going with the current license and all the invested time is "wasted". I have seen this so many times before and learned that my time is better invested somewhere else.

IANAL but you have 20 contributors to the project, if they have not transferred the copyright to you I guess you need their permission to change the license.

IANAL There is always a way around the current license, like for those who remembers qmail/djbdns where the user downloads and install the software and add patches.

@guysoft GPL-3 handles this in a good way i think. AGPL is a good choice when you want hosted work shareable.

guysoft, 2017-10-13:
@stone CraftBeerPi is transferred over wifi, and AGPL is actually something that covers that. Moreover some companies don't use software because its AGPL. It might have a better effect and do what @Manuel83 wants. It means if a commercial company tweaks it and uses it internally they have to release the source. AGPL means you have to release the source code if the program is transferred over a network. It means if people package and sell it.

Also, Gina pointed out -There at the moment misleading text in the README. It says ``CraftBeerPi is a free & open source project``.
That's misleading advertisement:
https://en.wikipedia.org/wiki/The_Open_Source_Definition#Definition
Namely:
3. Derived Works - The license must allow modifications and derived works, and must allow them to be distributed under the same terms as the license of the original software
5. No Discrimination Against Persons or Groups
6. No Discrimination Against Fields of Endeavor

But again, it really depends what @Manuel83 wants. I can only advise to open it, and perhaps consult with legal people because I can see there are several commercial projects already out there. And this one has potential to become a libre alternative.

guysoft, 2017-10-17:
Er - any progress @Manuel83 thoughts? What is missing for you to make a decision?

isarvalley, 2017-10-18:
So far, Manuel did a great job by adding a lot of new features. But as the project evolves it's getting more and more important to make it stable and reliable. In my opinion this is only possible by adding some code metrics, much more tests and so on. Not the fanciest thing to do but I guess this community has a lot of developer, like myself, which would be willing to do it. 

So it would be great if the licence could by changed into a real open source licence to motivate more developer to take part.  I hope there is one, which also takes Manuels whishes into account, which btw I can fully understand.

t04514219, 2017-10-18:
According to the GitHub guidelines you can legally fork the project. I also
would like to see the license change to a true open source license. I
rather help development on this project and not fork it.
5. License Grant to Other Users

Any User-Generated Content you post publicly, including issues, comments,
and contributions to other Users' repositories, may be viewed by others. By
setting your repositories to be viewed publicly, you agree to allow others
to view and "fork" your repositories (this means that others may make their
own copies of Content from your repositories in repositories they control).

If you set your pages and repositories to be viewed publicly, you grant
each User of GitHub a nonexclusive, worldwide license to use, display, and
perform Your Content through the GitHub Service and to reproduce Your
Content solely on GitHub as permitted through GitHub's functionality (for
example, through forking). You may grant further rights if you adopt a
license
<https://help.github.com/articles/adding-a-license-to-a-repository/#including-an-open-source-license-in-your-repository>.
If you are uploading Content you did not create or own, you are responsible
for ensuring that the Content you upload is licensed under terms that grant
these permissions to other GitHub Users.

aedocw, 2017-10-19:
Disclaimer: I am not a lawyer, what I'm writing here are merely my opinions.

The responses on this thread have been really helpful and productive in my opinion.  As a user (and small contributor) of CraftBeerPi getting this sorted out is pretty important to me.  Obviously I can continue using cbpi but I'm more concerned with keeping a strong community of contributors around it.

Given the commercial concerns @Manuel83 noted above, my suggestion would be to choose a strong copyleft license for this project.  If you want to maintain an open source friendly license it's very difficult to also limit commercial use.  It's possible[1] but reasonably difficult IMO.

You can find more details on OSI licenses here[2], and this writeup[3] gives you a pretty easy to read overview of concerns around open source licenses.  Finally, this faq[4] has some good answers around limits you can impose.

It does seem that if you really want to have a healthy community of contributors, you also run the risk of seeing someone commercialize it.  At a minimum though you can make it problematic for them to do so without maintaining license compliance (and if they are not in compliance and you feel like going through the effort, you can probably take them to court - that fact should discourage anyone who was just hoping to make a quick buck off the project).

[1]: http://www.gnu.org/philosophy/selling-exceptions.en.html
[2]: https://opensource.org/licenses/category
[3]: https://opensource.com/law/13/1/which-open-source-software-license-should-i-use
[4]: https://opensource.org/faq

guysoft, 2017-10-21:
So just to note - there is someone that is quite clearly violating the license and is advertising his product with craftbeerpi in the official Facebook group. So this license is not only blocking me from distributing a free open souce distro of craftbeerpi, its also not stopping commercial sellers to redistribute and do exactly what @Manuel83 is trying to stop.

Link:
https://m.facebook.com/groups/1620423084927502?view=permalink&id=1735864836716659

Screenshot if you have no access:
[image]
up:content://media/external/file/109105

guysoft, 2017-11-21:
Sent @Manuel83 and email about a month a go, no answer :(

oleost, 2017-12-02:
@Manuel83 You clearly should come with an statement.

As a community software Craftbeerpi seems to be hurting from the license, and it's getting hold back.

D4p0up, 2018-05-29:
Hey @Manuel83, seems like you're putting some effort today on CBP3 GUI. Kudos.

Upon next release, would you mind reconsidering this license switch?

Thanks!

rwshieldsiii, 2019-02-10:
@Manuel83 any update?  I’d love to contribute to this as well!

linuxguy123, 2019-02-13:
I am really frustrated that this software isn't true open source.  I get that Manuel doesn't want his work taken advantage of commercially, but guess what... when you open source software, anyone can use it for anything.  If you aren't comfortable with that, don't make your software open source.

What I don't get is why we aren't developing a replacement for CBPi ?

linuxguy123, 2019-02-13:
FYI, the same sort of thing happened with QCad.  Finally the open source community took over and developed libreCAD.

guysoft, 2019-02-13:
Manuel also is against derIvative work. I released CraftBeerPiOS which is
build script which generates an image ready to flash on a raspberrypi,
similar to OctoPi.
And he told me to take off the git repo of thr build script, even though it
does not violate the license.

On the other hand, some users on the Facebook group publicly state they
sell hardware with craftbeerpi pre-installed. Which is against the license
and also its the reason Manuel claims he is not opening the source.
So I gave up further development.

linuxguy123, 2019-02-13:
It is time for us to build a clone/replacement for CBPi.

I built a bit of a class framework last night.  Just simple stuff like Brewstand, Vessel, Controller, Sensor and Actor.

A Brewstand has one or more Vessels.
A Vessel has one or more Controllers.
A Controller has one or more Sensors and controls one Actor.
Sensors have type temperature, flow, weight, etc.
Actors have type heater, pump, mixer.

To add a controller, you superclass Controller like
class myPID (Controller):
        <your code here>

To add a sensor, you superclass Sensor like
class myDS18B20(Sensor)
        <your code here>

To add an actor, you superclass Actor like
class myElement(Actor)
       <your code here>

Everything is in Python3.

I am thinking I'll develop a framework and run it as a console app (process) first and then add a flask app on top that communicates with the framework.  That way if someone wants to build something different, they can still use the framework.
  
Wrong approach ?  Feedback ?

archonic, 2019-02-13:
@linuxguy123 It's definitely time for those of us that can code to get off our butts and make something we want to see in the world. I've made a few starts over the last couple years. I've come to the conclusion that requiring a router to brew is silly. I'd personally like to see a local solution using the rpi touchscreen, so I like the console app approach. I'm happy to build a local solution on top of that process, probably using TkInter.

linuxguy123, 2019-02-13:
@archonic   I love that CBPi uses a router and is accessible from multiple browsers.  FYI, you can easily set an RPi up to an access point by itself, no router required.

Your desire to build something different than what I want just goes to prove that building a framework is a good idea.

linuxguy123, 2019-02-13:
FYI, discussion is going on here: https://github.com/Manuel83/craftbeerpi3/issues/196

KnoAll, 2019-02-13:
Has anyone looked at BrewBlox? How does that compare with CBPi?
https://brewblox.netlify.com/

guysoft, 2019-02-14:
@KnoAll Er, looks cute, not sure the features are the same. Its written in Node.
I could try and build a raspberrypi image to it similar to what I have done with [CraftBeerPiOS](https://github.com/guysoft/CraftBeerPiOS) using the docker module I have there.

linuxguy123, 2019-02-14:
What is the license on brewblox ?     Is it related to BrewPi ?

KnoAll, 2019-02-14:
Yes, it is all related. Brewpi is the fermentation piece and i think brewblox is is the brewing piece. as far as i know it is some kind of GNU license.

linuxguy123, 2019-02-14:
Double check that license because BrewPi has some sort of "contribution agreement".   Also, if you check the BrewBlox repositories, there is no license file in some of the directories.   There is in some, but not in others.

Furthermore, I don't know exactly what BrewBlox does, but it looks pretty complicated.

linuxguy123, 2019-02-16:
How was the web UI done in CraftBeerPi ?  What tool was used to generate it ?

D4p0up, 2019-02-16:
Front the Blueprint import in the UI folder I guess this is some Flask implementation.

I'm really curious to what a real open source alternative could be. I've been thinking about alternative solutions architectures that would make it easy to maintain/evoluate, and came across this : https://auth0.com/blog/reactive-programming-in-python/

That looks like a good alternative to classical Method calling paradigm, for the backend and might be a good starting point.

guysoft, 2020-11-13:
Hey, now that craftbeerpi 4 is a dead project - can we open source craftbeer 2?