craftbeerpi/craftbeerpi #72: Prevent empty kettle name in setup

MrLeeh, 2016-08-14:
Prevent the usage of an empty Kettle name field in the basic setup. Otherwise you won' t be able to modify the kettle later on as there is no link text to click on. You should check for a string length > 1 here. 

This isn' t a problem if you add a kettle in the "Kettle" menu. Here you already check for an empty name. Although it would be nice to get a message that the name is missing.

Manuel83, 2016-08-14:
Fixed in Version 2.2.

MrLeeh, 2016-08-14:
Just pulled the dev Version 2.2 and the error persists:

[image]

Manuel83, 2016-08-14:
You create the kettle during setup wizard right? That's what I can reproduce. 
If you create a kettle later it should not be possible to save the kettle with an empty name. 

Deactivated save button
[image]

Activated save button
[image]

MrLeeh, 2016-08-14:
Yes it is during setup wizard. Later it' s not possible to save the kettle.

gimiki, 2017-01-18:
There is the same bug on "hardware configuration":
I can add any device with *no name* even if the save button is deactivated.

Mrpond1970, 2017-01-19:
I found the same problem a while ago

 https://github.com/Manuel83/craftbeerpi/issues/98