craftbeerpi/craftbeerpi #74: adding hardware

paalja, 2016-08-17:
hi
this goes for the latest dev2.2:

what I setup in the initial startup config is not quite what I see later

At this config - look at the thermometers and pump4

**hw setup**
[image]

**kettle setup**
[image]

**Hardware tab**
the thermometer definition is not in the hardware tab
[image]

**Dashboard**
No thermometers on the dashboard but 4 of them are linked to the kettles.
Pump 4 still missing
[image]

**Pump 4 appears**
Pump 4 appears when I delete Pump 1 in the hardware tab
[image]

Manuel83, 2016-08-17:
You have really a big setup. :-)

I fixed the issue. Please retest.

paalja, 2016-08-18:
We're building a HERMS, and this is not as big as it might seems, misleading description :)
There will be 3 elements/relays for the HLT, the same for the boiler, and one element/SSR for the HX.

Seems to be working fine now, thanks alot again and great work!! :+1: 

[image]

closed 2016-08-18 by Manuel83

RenatoTana, 2016-08-21:
Dear paalja,
Could you please tell how did you connected the thermometers to the Raspberry Pi? Do you have any diagram?

paalja, 2016-08-21:
hi,

I have no diagram but you'll find many good examples of how to do that. Below is three pictures from the front and back of the bus of my setup

[image]

[image]

[image]

RenatoTana, 2016-08-21:
Thank you very much for the pictures. Very nice assembly of yours!
My questions was more on the way to where to wire the sensors at the Pi? Should I use a regular GPIO for it? Because on the configuration is not possible to choose a GPIO for the thermometers. I have 3 x DS1820, but no clue on where to wire them at the Raspberry Pi.

owold, 2016-08-21:
Suggest doing a simple google search before asking

https://www.google.no/search?q=connecting+temperature+sensor+to+raspberry+pi&biw=1467&bih=1295&source=lnms&tbm=isch&sa=X&sqi=2&ved=0ahUKEwil_dHHmdPOAhWBrSwKHfb9AfUQ_AUIBigB

oj