craftbeerpi/craftbeerpi #75: Installation dev2.2 - Flask-SocketIO conflict

RenatoTana, 2016-08-19:
Dear Manuel,
I`m having an issue when I`m trying to install de version dev2.2 over a pre-installed 2.1 and I try to run the server (sudo python runserver.py). There is some conflict with the version of the Flask-SocketIO and I`m not able to run the server. (receiving an error message)
Question, is it possible to install the version dev2.2 from a just installed Raspibian? I mean, without installing first the version 2.1? If it is so, could you please tell how?
Using a Raspiberry Pi B

Manuel83, 2016-08-19:
Please try to install v2.2 on a clean Raspbian installation. In version 2.2 I did a big lib update.
But before you do this it would be nice if you could post the error message.

tomandre76, 2016-08-19:
what´s the code for installing 2.2 ?

RenatoTana, 2016-08-19:
Thanks for your answer Manuel!
And Thanks tomandre76, this would be my question. How to install the 2.2 on a fresh Raspibian?

Manuel, Sorry but I've formated already the SD card and installed a fresh Raspbian. The error was about the Flask-SocketIO, it says that this was an old version and asked to uninstall and install a new one python-socketio.
Next time I gonna save the error message

Manuel83, 2016-08-19:
Directly at the start of Raspbian you can hold the "shift" key for recovery. That's the fastest way to have clan OS.

To install v2.2. just type:

git clone -b dev2.2 https://github.com/Manuel83/craftbeerpi

After this run the install.sh

RenatoTana, 2016-08-20:
Thank you. The instalation suceed.

closed 2016-08-20 by RenatoTana