craftbeerpi/craftbeerpi #76: feature request : group objects / add more objects to one kettle

paalja, 2016-08-21:
Is it possible to add a feature which can make the 'heater's  as an object and add them to a group, and the add the group to a kettle?
Or just add several objects/heaters/relays to one kettle?

In my setup I have three 3,5kW elements in the HLT and in the boiler. 
I have one SSR controlling each element because of the high current of ~15Amps.

Currently I would be able to attach only one element to one kettle.

I could of course attach the three SSRs to one GPIO but I am afraid I would exceed the max current of the GPIO of max ~50mA

[image]

[image]

Blom1974, 2016-08-22:
If your main worry is powerconsumption on the port i suggest you build an driver stage for the output.
I would recommend that anyway beacuse it is borderline that the ouputs have enough power to activate an SSR.

**I might be as simple as this:**
[image]

**Something like:**
R2 = 1k (optional)
R1 = 10k
LOAD = SSR
+V = Suggest  5V from Pi

Gpio pin is connected to the Drive signal pin that goes to R2

Blom1974, 2016-08-22:
With that said the grouping function or be able to assign more than one output to the same function could be nice to have.