craftbeerpi/craftbeerpi #78: More than one recipe at the same time

Eremitos, 2016-08-23:
Private brewers have sometimes more than one kettle.
To save time it would be helpfull to start more than one recipe at the same time and work on different kettles parallel.
Is it possible to assign a recipe to a specific kettle instead of using all the kettles to one recipe?
We already discussed this topic on hobbybrauer.de.

[image]

Manuel83, 2016-08-24:
I'm currently working on the fermentation module. For this module you can have multiple processes. If this is working well I can think about using the same concept for brewing.

[image]

Eremitos, 2016-08-29:
Hello Manuel,
is this fermentation module already in Dev 2.2 or how can I use it?

Manuel83, 2016-08-29:
Some parts are in but it's not ready for use now. The last missing piece is the step control. Hopefully ready soon. Regards Manuel