craftbeerpi/craftbeerpi #80: Interface to integrate other sensor systems

olemde, 2016-09-02:
Hi Manuel
i think it would be nice if users like me who are not so familar in python programming are able to integrate other sensor systems into the CBP. 
I am able to program an arduino board in C, but the webserver and flask stuff is to complex for me. My idea is that the arduino sends values in a structured format via a serial line to the raspberry. 
Examples could be a ultrasonic level measurement of the kettle.

Manuel83, 2016-09-02:
The question is what would you like to do with the sensor data? Just display it on screen?
Flask is quite simple but the AngularJS JavaScript part is a little bit more complex.
I would assume that connecting a ultrasonic sensor direct to the PI is less complex.

http://www.raspberrypi-spy.co.uk/2012/12/ultrasonic-distance-measurement-using-python-part-1/

You just have to write one method like the following 

@brewjob(key="readdistance", interval=5)
def readDistance():
   ## HERE YOU ADD THE utrasonic code

olemde, 2016-09-02:
To display it would be the first step. The main target is to control the sparging in the lauter tun.