craftbeerpi/craftbeerpi #83: new boiling logic - target temp 98° + heater always on

franktegtmeyer, 2016-09-04:
please add a seperate boiling logic:
- make the temperature, the timer starts running configurable (to allow e.g. 98°)
- make the heater be always on while in boiling logic

denemcd, 2016-12-23:
I'm just trying to work on some boil logic and I can't seem to make this work. After some discussion on Facebook with Manuel I've come here. My theory is that this should work but I can't seem to get it to work,

```
from automaticlogic import *
from brewapp import app
import time
from brewapp.base.util import *
from brewapp import app, socketio

@brewautomatic()
class BoilLogic(Automatic):

    state = False

    def run(self):

        while self.isRunning():
            self.state = True
            self.switchHeaterON()
            socketio.sleep(1)
```

denemcd, 2016-12-23:
Have even tried checking to see if the it's running and only turning it on when not and it still doesn't work

[from automaticlogic import *
from brewapp import app
import time
from brewapp.base.util import *
from brewapp import app, socketio

@brewautomatic()
class BoilLogic(Automatic):

    ## Define config paramter as array of dicts
    # configparameter = [{"name":"Overshoot","value": 0}]

    state = False

    def run(self):

        while self.isRunning():

            ## If element is off turn on
            if(self.state == True):
                self.state = True
                self.switchHeaterON()

        socketio.sleep(1)](url)

Manuel83, 2016-12-24:
Hi

First of all I discovered that every logic needs a config parameter. Otherwise an exception is thrown on server side. This is a bug I have to fix. As workaround just add an empty parameter you are not using. 

The `socketio.sleep(1)` in your code has a wrong code indentation.
Thats very important at python. It was not part of the while loop.
Furthermore make sure that you switch off the heater when stopping the automatic. 

I updated your code:
```
from automaticlogic import *
from brewapp import app
import time
from brewapp.base.util import *
from brewapp import app, socketio

@brewautomatic()
class BoilLogic(Automatic):

    ## Define config paramter as array of dicts
    configparameter = [{"name":"Boil Temp","value": 99}]

    state = False

    def run(self):

        while self.isRunning():

            ## If element is off turn on
            if(self.state == False):
                self.state = True
                self.switchHeaterON()

            socketio.sleep(1)

        ##Switch off when finished
        self.switchHeaterOFF()
```

Merry Christmas

denemcd, 2016-12-24:
This is excellent. Tested and it works. I should have realised the sleep was in the wrong place. It would be useful to have this in the code going forward as there will be more people wanting this for BIAB and all in one systems in the future.

Will send something via pay pal in the next few days for you to get a coffee or beer.

xavanty, 2017-01-03:
Where do I install this new logic so I can use it?

ivanbate, 2017-05-30:
If i just copy this code, create a .py file and place in the brewapp/base folder will this logic appear in the logic list on kettles. I like the idea of this feature.