craftbeerpi/craftbeerpi #86: Temperatura Update

pjaragao, 2016-09-12:
The temperature don't update, I tried Safari, Firefox and Chrome, need to refresh to get the temperature updated. (OS: Mac OS X 10.11.6)

Manuel83, 2016-09-12:
Can you provide the app.log (root folder of CBP) and a Screenshot of your temperature config please.

pjaragao, 2016-09-12:
tomorrow I send you, left on the office and the VPN is off.

regards,

pjaragao, 2016-09-13:
https://www.dropbox.com/sh/ik2k72p3h8zl2pk/AABFKNA_rSn3xTwYeRNNBG6Pa?dl=0

I created a folder with app.log and some screenshots, if you need something else let me know. Great app, by the way.

regards,

Manuel83, 2016-09-14:
App.log looks ok. Can you try to user Chome and open the JavaScript Console.

On Windows press F12 on Mac it's ALT+CMD+I

Check the tab console in the Chrome Developer extension. Is there any JavaScript error?

pjaragao, 2016-09-14:
I'm doing a fresh install of raspbian jessie lite and CBP to test again. Let you know ASAP.

pjaragao, 2016-09-14:
After the fresh install everything works fine. But now I have another doubt, after add the relay GPIO for heater at kettle and pump and save the relay turn on, but at the dashboard the status of the heaters and relays are OFF (button in grey), when I press to turn ON(button green) the relay turn off. Is that correct?

pjaragao, 2016-09-14:
realized that the temperature stops to update when the is one or more kettles without thermometer assigned.

Manuel83, 2016-09-18:
I will check this.

bluetaz, 2017-04-16:
i'm having also the problem that the temperature doesn't auomaticly aupdates. Is there a solution?