craftbeerpi/craftbeerpi #88: Low priority suggestion: LOGO!

MrDirtt, 2016-09-17:
One thing that alot of people seem to miss from Strangebrew is the ability to have your brewery logo on the Dashboard.  Your system seems to be more dynamic and you already have a place to adjust variables such as brewery name...so I was thinking this would be an excellent spot to add a fun feature like Logo.

Manuel83, 2016-09-18:
Nice Idea. I'm thinking about the right place on screen.
Any suggestion is welcome.

Blom1974, 2016-09-18:
Top left corner would be nice. But i think that your logo should be somewhere well visible as well.
For ex centered in the bottom as a " Powered by CraftbeerPi" and your logo.

archonic, 2017-04-03:
I'd argue that a logo and even a small "powered by" any place on the interface doesn't help you brew beer, and shouldn't be a part of that layout. It's already got a splash screen for language selection - that's the perfect place for it.