craftbeerpi/craftbeerpi #90: Sensor showing wrong values

Nauticno, 2016-09-20:
[image]
Sensor showing wrong values. Looks like F but I set ut C. I started With one sensor. When adding a new it started. Tried to remove craftbeerpi.db and restart with no luck.

What is happening?

Nauticno, 2016-09-20:
[image]
The temperatur from "Biltema" is going up and down. From about 27 degree celsius til 85 degress celsius.

Manuel83, 2016-09-20:
Seams that your sensor is sending wrong data. 

Common Sensor errors
```
 0	 = data line or GND missing
 85	 = internal sensor error, maybe the cable is too long
 -127 = no supply voltage (missing VCC) 
```

How is you sensor connected?
You can check the basic sensor file on the Raspberry Pi

```
cd /sys/bus/w1/devices
cd 10-000801b5*
cat w1_slave

0f 00 4b 46 ff ff 06 10 0c : crc=0c YES
0f 00 4b 46 ff ff 06 10 0c t=7375 
```

Nauticno, 2016-09-20:
Thank you for quick replay!

My values:
pi@raspberrypi:/sys/bus/w1/devices/28-04166183d2ff $ cat w1_slave
9b 01 4b 46 7f ff 0c 10 dc : crc=dc YES
9b 01 4b 46 7f ff 0c 10 dc t=25687

I'm not sure what this means?
I shall check my connections. Try to change to Wago connectors tomorrow,

Manuel83, 2016-09-20:
9b 01 4b 46 7f ff 0c 10 dc : crc=dc YES
9b 01 4b 46 7f ff 0c 10 dc t=**25687**

That means 25.687 °C

The question is why your sensor is sending a error value of 85 so often.

Nauticno, 2016-10-29:
Perhaps because itis bougt from Ebay? http://www.ebay.com/itm/281725483259?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT 
Sometimes it works, other time it don't. Error 0 , meaning no Connection....

Nauticno, 2016-11-07:
After reconnect by soldier cables together it works!

closed 2016-11-08 by Nauticno