craftbeerpi/craftbeerpi #91: Temperature Porb not recognized

Denismarco, 2016-09-29:
Hi, 

I'm trying to setup the teperature probe but even if I do exactly the same as blueprint, I still got "no thermometer" in the kettle board... 

Do I have to do something else to activate it?

Nauticno, 2016-11-07:
I had same problem. Make sure connection are 100% When I soldier them together.

rlcout, 2016-12-25:
I had this problem too.  And a have a card with carftbeerpi 2.1 and works fine, but when a change the card to version 2.2 olny shows dummy probe. Anyone knows what's wrong?

rlcout, 2016-12-30:
i reinstall de system again and works fine now

FuzzyWaffler, 2017-06-28:
I'm on a new one and install and I'm having the same issues. They appear as dummy and not as one wire. When I use dummy it only shows a static temp of 68f

FuzzyWaffler, 2017-06-29:
My issue was me. I soldered the wires so they bypasses the resistor. Facepalm...

Resolved.