craftbeerpi/craftbeerpi #95: Buzzer Always on ? Possible to add invert option

mikehickey, 2016-10-09:
Dear manuel thank you for developing such a fantastic bit of software
I love how you can so easily customise 2.2 to brewing and fermentation
I have an "active" buzzer set to gpio 20 on my setup
It is currently constantly on and beeps "off" at the end of a brew step
Is it possible to add an invert option to the configuration of the buzzer ?
I have tried different gpio positions so I do not think it is a case of the gpio booting to a high state . If I set the buzzer as a pump inverted it works perfectly

Many thanks

Mike 

Current set up
Electric herms
3 vessels 2 pumps
Two Ssr and 4 relays
Kegerator with heating  and chilling
Rip b +

Ridemywideglide, 2017-04-03:
My temp band-aid for this problem was to edit the buzzer.py file, and look for the lines "sound1" "sound2" "sound3", and reverse the H's and L's..  Change all H's to L's, and all L's to H's. But this does not stop the problem of the buzzer coming on every time you startup the pi. It will stay on until you hit "start". I just hit start then stop real quick at the moment, but I don't have this hooked up to any heaters yet. I see this being a problem when it's actually firing a heater.

archonic, 2017-04-03:
There's an option in the interface to invert heaters. As long as you know how your SSRs trigger, it shouldn't be a problem. Covering elements with water would be a good thing to do on first time setup if you're not sure. Adding the option to invert a buzzer sounds sensible.

Ridemywideglide, 2017-04-07:
The problem is, even with reversing it I posted above, the buzzer is still on every time you startup the pi... Stays on until you cycle a start-stop operation.. It may be nice to have it pulse when it's up, but not stay on..

JamFfm, 2017-04-09:
http://www.ebay.de/itm/141648178257?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

This one solved my Problem. I first tried it with am active buzzer. Same Problems as discribed above.

I tried it with soldering the 2 poles of the soundelement connecting with GND and GPIO (16). That is not loud enough

mikehickey, 2017-04-10:
Dear all

It seems on the raspberry pi holds the gpio you nominate to control your buzzer at +5v. When buzzing it momentarily takes the gpio to ground opposite of what you would expect. Not sure if this is the software or the way the raspberry pi sets up the gpio pins on boot up.
I solved this one by connecting together the  +5v and the vcc lead on my active buzzer. These 2 leads were then conneted to +5v pin on the raspberry pi. I then connected the ground wire of my active buzzer to the control Gpio. Therefore in the resting  state all 3 wires are held at 5v volts so nothing happens. When craftbeerpi beeps the gpio is taken to ground three times I therefore get three beeps. Worked for me 

Happy brewing craftbeerpi is fantastic !
All the best mike

lexo77, 2017-04-28:
Hi,
I had the same problem, modifying the file *\brewapp\base\buzzer.py solved it.
1. set GPIO to HIGH in initBuzzer()
2. switch H <--> L in soundX

```

def initBuzzer():

    buzzer_gpio = app.brewapp_config.get("BUZZER_GPIO", None)
    app.logger.info("BUZZER GPIO: " + str(buzzer_gpio) )
    try:
        if buzzer_gpio is not None:
            buzzer_gpio = int(buzzer_gpio)
        GPIO.setmode(GPIO.BCM)
        #GPIO.setup(buzzer_gpio, GPIO.IN)
        GPIO.setup(buzzer_gpio, GPIO.OUT)
        GPIO.output(int(buzzer_gpio),GPIO.HIGH)

    except Exception as e:
        app.logger.error(e)

def nextStepBeep():
    start_new_thread(playSound,(sound1,))

def timerBeep():
    start_new_thread(playSound,(sound2,))

def resetBeep():
    start_new_thread(playSound,(sound3,))

## Melodie Pattern
## H = HIGH
## L = LOW
## Float value as pause
## it must be a L at the end to turn the sound off
sound1 = ["L",0.1,"H",0.1,"L",0.1,"H",0.1,"L",0.1,"H"]
sound2 = ["L",0.1,"H",0.1,"L",0.1,"H",0.1,"L",0.1,"H"]
sound3 = ["L",0.1,"H",0.1,"L",0.1,"H",0.1,"L",0.1,"H"]
```

kostasvit, 2017-06-14:
Hello,

I have the same issue with mikehickey. It keeps beeping constantly when i plug in the wires. Especially when i connect the ground and the power is already connected. It is weird because i tested on a breadboard i had the same issues and then somehow it worked perfectly. Nothing happens when i connect the gpio cable. What is <<vcc>> that mikehickey mention? Is there any other way to fix that so it would work properly? Nothing fixed with lexo77's way.

Many thanks

el-lolo, 2019-09-20:
hi 
I have the same problem but I don t find the  buzzer.py 
Anyone can help me please?