craftbeerpi/craftbeerpi #96: some bugs in 2.2

tkranz, 2016-10-12:
Hi Manuel,

first of all very nice work !!! i developed the same a few years ago for Arduion "Brauknecht" :-)
Here a list of issues i found in 2.2 beta:

1. If you stop the brewing process the target Temp should be set to 0 but it isn't.
2. I have the problem that it doesn't show the "controlllogic" if i edit the vessel always "please select"
3. The temp cart is not working any more, just shows calculation/loading sign.

Thanks
Thomas

tkranz, 2016-10-12:
one more,
if you import new recipe the name of the old recipe stays on top

Manuel83, 2016-10-12:
Hi,

your first point is a good idea. For the other point I will have look. Can't be a big issue.

Thanks for your feedback.

kapau774, 2017-04-20:
Hi everyone,

Addition to 3)
I had the same problem which occurs if the raspberry pi is not shut down and only disconnected from power supply. The templog-file contains some "null" values which cannot read by strptime:
```

ERROR:brewapp:Exception on /api/temp/K/1/chart [GET]
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/craftbeerpi/brewapp/base/thermo.py", line 94, in temp_chart
    result = {"name": name, "data": read_temp_log('./log/' + type + '_' + str(id) + '.templog')}
  File "/home/pi/craftbeerpi/brewapp/base/util.py", line 156, in read_temp_log
    time = int((datetime.datetime.strptime(row[0], "%Y-%m-%d %H:%M:%S") - datetime.datetime(1970, 1, 1)).total_seconds()) * 1000
  File "/usr/lib/python2.7/_strptime.py", line 325, in _strptime
    (data_string, format))
ValueError: time data '2017-04-2017-04-11 17:17:17' does not match format '%Y-%m-%d %H:%M:%S'
```

Deleting the values within the *.templog file solves the problem (or shutting down the pi properly ;)).  

One suggestion would be to change the code within the util.py (from line 153) to

   ```
 data = f.readlines()

    for line in data:
        line = line.replace("\x00", "").replace("\n", "").split(",") 
        time = int((datetime.datetime.strptime(line[0], "%Y-%m-%d %H:%M:%S")
                    - datetime.datetime(1970, 1, 1)).total_seconds()) * 1000
```

which replaces all the null values before applying the strptime.