craftbeerpi/craftbeerpi #99: Support for flow meters

h44r3k, 2016-10-24:
What about adding support for flow meters so that you can just type in the volume you need for strike and sparge water in CBP? This would be awesome if you´re following the brew steps in BeerSmith. For example, if you need 20L of water for sparging, the flow meter will shut off the pump when the desired volume is reached.

blazem, 2016-11-22:
This is a cool idea. What flow meters would be a good choice?

h44r3k, 2016-12-11:
I´ve ordered a couple of these to play with: http://www.ebay.co.uk/itm/Water-Flow-Sensor-1-2-Fluid-Flowmeter-Switch-Counter-1-30L-min-Meter-NEW-/262450179287?hash=item3d1b3fecd7:g:ZzoAAOSwH6lXQqdy. They will work with 5V from the RPi. I´m currently trying to find some code for RPi to start from.. There are many working solutions for arduino and kombos with RPi but few for RPi only.

blazem, 2016-12-11:
> Water-Flow-Sensor-1-2-Fluid-Flowmeter-Switch-Counter-1-30L-min-Meter-NEW-/
> 262450179287?hash=item3d1b3fecd7:g:ZzoAAOSwH6lXQqdy. They will work with

Keep me posted, I would like to see something like this implemented on my
system at some stage.

mjg964, 2017-10-23:
It would be awesome to integrate a "dispensing" tab in CraftBeerPi and use these in a kegerator for keg volume monitoring! Could this be adapted? https://learn.adafruit.com/adafruit-keg-bot?view=all

blucmal, 2018-07-25:
With the flow meter is there anyway to use it as a element kill switch for a rims/herms setup so it turns off  the element when there is no liquid flow in the case of a stuck sparge?

blucmal, 2018-07-28:
Should have sent stuck lauter cheers