craftbeerpi/craftbeerpi #104: CraftBeerPi 2.2, installation

Nauticno, 2016-11-13:
I have just installed version 2.2 by "git clone -b dev2.2 https://github.com/Manuel83/craftbeerpi"
It ask me to start setup, Setting Brewery Name
Select GPIO, Select 1Wire, Add Heater. add thermometer. Add kettle. Let's go brewing, and then back to start. Looping forever. 

Why?

wvdbroek, 2016-11-15:
same problem here did you find a solution yet ?

Manuel83, 2016-11-15:
Which browser are you using?

wvdbroek, 2016-11-15:
Internet Explorer but it seems resolved after a restart of the pi and a manual restart of the setup

wvdbroek, 2016-11-15:
I did a Clean instalation of the rasbian and a Clean one of the 2.2 Version. I mean i Have only 2.2 Running No 2.1

Nauticno, 2016-12-02:
I installed 2.1. Then I removed 2.1 by

- sudo rm /etc/init.d/craftbeerpiboot
- sudo rm -rf /home/pi/craftbeerpi
Then install by git clone -b dev2.2 https://github.com/Manuel83/craftbeerpi
Then I got "The gevent-socketio package is incompatible with this version of the Flask-SocketIO extension. Please uninstall it, and then install the latest version of python-socketio in its place."
Solution for this:

- sudo pip uninstall flask-socketio
- sudo pip install flask-socketio

Then it worked!