craftbeerpi/craftbeerpi #106: Can't get craftbeerpi to install right.

ulbas, 2016-11-18:
Hi i've tried for some days now to get my craftbeerpi to run. But it seems to Get wrong when i have downloaded and installed it and i try to run the runserver.py. Then i get the message error 98 server address 0.0.0.0 5000 is already in use. Anybody here that have a solution that could help me. I've tried to install both versjon 2.1 and 2.2. And i have re installed the OS and tried everything from scratch

johnzan, 2016-11-25:
Hi, not sure how much help this will be, but that message indicates that either the app is already running OR something else is running and listening on that port number. 

If you run the following command:

`ps -ef | grep python`

you'll see if the runserver.py command is running

hope this helps.
Rgds.,
J.

ulbas, 2016-11-25:
Thx, I found out the problem a couple of days ago but forgot to take down the post. The problem was that the server was allready running and that was the problem with why i didnt get to run the runserver.py file. But thanks anyway :)

closed 2016-11-28 by Manuel83