craftbeerpi/craftbeerpi #110: Version 2.2: Die Temperaturgraphik dauert lange beim aufruf

JamFfm, 2016-11-28:
Wenn man die Temperaturgraphen aufruft brauchen diese sehr lange bis sie angezeigt werden. Manchmal hilft nur löschen und dann erneut aufrufen. 
Klasse Projekt. 
Schade das ich nicht in Mainz sein könnte.

Manuel83, 2016-11-28:
Hi,
wie viele Einträge hat dein Templog. Bitte schau mal unter /home/pi/craftbeerp/log
Da müssten .templog files liegen. Die Anzahl Zeilen wäre interessant.

Gruß,
Manuel

JamFfm, 2016-11-29:
Ich habe in der Datei F_1.templog 4859 Zeilen, K_1.templog 45410 Zeilen, app.log 111160 Zeilen.
Habe 2 Kettle mit zugewiesenen Temperatursensoren. Der erste Graph ist nach ca. 14 sec da , der 2 Graph >3 min habe dann abgebrochen. 
Dann im 2. Graph die Daten mit dem Button gelöscht und siehe da dann ist der Graph innerhalb 2 Sec da. In Nano sehe ich dann auch gerade mal 2 Zeilen. 
Die Anzeigezeit ist also abhängig von der Anzahl der Zeilen. Evtl könnte hier ein Loop eingebaut werden so das die Daten älter als 24h oder 48 h gelöscht werden (round Robin).

JamFfm, 2016-12-21:
Mit der neuen Version ist die Anzeige erheblich schneller. Lasse es noch einige Daten sammeln.

JamFfm, 2016-12-23:
ja ist erheblich schneller. Danke!!

closed 2016-12-23 by JamFfm