craftbeerpi/craftbeerpi #113: Dev 2.2 Graph display issue

ivanbate, 2016-12-03:
Hi
I have noticed i am unable to display the graph of temperatures in dev 2.2. Not sure if this is feature still under dev or there is a bug.
I receive the spinning refresh arrows on the screen after loading the graph but the graph image does not appear.

Thanks
Ivan

JamFfm, 2016-12-04:
Try to delete the graph then back to brewing. Now it should be possible to display the graph. 
The ammount of Lines is the Problem.

Manuel83, 2016-12-06:
Hi 

are you using the graph for a brewing kettle or for a fermenter?
I'm currently improving the implementation.

Regards,
Manuel

Manuel83, 2016-12-24:
The version2.2 got new chart implementation.

closed 2016-12-24 by Manuel83

JustBos, 2016-12-30:
I'm running 2.2 and only using as fermentation but I am also having issues with the chart displaying. Deleting does not seem to work, when trying to download I get a not found error message. 

Cheers

duzvik, 2016-12-31:
try to remove K_1.templog from disk.
I app.log there was error:
ERROR:brewapp:Exception on /api/temp/K/1/chart [GET]
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/craftbeerpi/brewapp/base/thermo.py", line 94, in temp_chart
    result = {"name": name, "data": read_temp_log('./log/' + type + '_' + str(id) + '.templog')}
  File "/home/pi/craftbeerpi/brewapp/base/util.py", line 154, in read_temp_log
    for row in reader:
Error: line contains NULL byte

JustBos, 2017-01-01:
@duzvik that worked, thankyou. 

I deleted both F_1.templog and F_2.templog as I have 2 fermenters setup.

Working now.
Cheers.

guysoft, 2017-10-13:
Still getting this bug in 2.2:
```
ERROR:brewapp:Exception on /api/temp/K/1/chart [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1988, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1641, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1544, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1639, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1625, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/craftbeerpi/brewapp/base/thermo.py", line 94, in temp_chart
    result = {"name": name, "data": read_temp_log('./log/' + type + '_' + str(id) + '.templog')}
  File "/home/pi/craftbeerpi/brewapp/base/util.py", line 154, in read_temp_log
    for row in reader:
Error: line contains NULL byte
```

The line is here:

```
def read_temp_log(file):
    import csv
    result = {"temp": [], "target_temp": []}

    if os.path.isfile(file) == False:
        return result
    with open(file, 'rb') as f:
        reader = csv.reader(f)
        for row in reader:
            #print row
            time = int((datetime.datetime.strptime(row[0], "%Y-%m-%d %H:%M:%S") - datetime.datetime(1970, 1, 1)).total_seconds()) * 1000
            result["temp"].append([time, float(row[1])])

            result["target_temp"].append([time, float(row[2])])
    return result
```
Workaround to delete the temp works. 

Solution would be to replace this line:

```
    with open(file, 'rb') as f:
```

With this:

```
import codecs
    with open(codecs.open(file, 'rU', 'utf-16'), 'rb') as f:
```

[solution taken from here](https://stackoverflow.com/questions/7894856/line-contains-null-byte-in-csv-reader-python).
Would pull request when we solve #67