craftbeerpi/craftbeerpi #114: BeerBrew 30 version 2 with 2 heating elements

Nauticno, 2016-12-06:
I have an old BB30 with 2 buttons. One for 1000w and one for 1500w. These buttons are a litle loose so a played with a idea using 2 SSR relays instead of the buttons. Then the PID logic  must be rewritten so it activates 2500w to start the heating process. And then use 1000w for keeping right temperature. The BB30 must be modified by an electrican specialist!

This can be used for other kettles to.

Is this possible to consider as an new  feature?

Manuel83, 2016-12-06:
Yes could be a new feature for the version 2.3.

chriszero, 2017-03-04:
I've building a kettle with the same configuration, two heaters. Separately switchable.
I would implement it the same way as a heating cascade for buildings.
- Start heating with first heater
- If temperature delta is lower than X in time T, activate second heater.
- if temperature delta is higher than X in the T, deactivate the second heater, reset I of PID for first heater.
- PID logic only on first heater.

This would help for rapid heatup, for just holding the temperature, one element should be enough. If booth elements are needed to hold the temperature, hey should be used in parallel. So no need to control them separately.

I need this feature "Now". ;-) I implement the feature and end you a pull request if it's ready.

kjartaneglodden, 2017-03-08:
Hi

I don't really need PID at all (I think), but I have the following issue/setup:
**HLT**
- 2500W x 3 heating elements engaged by contactors
- Pump with valves(actuated) for Recirculating & transfer to mash tank

**Mash tun**
- No heat, just insulation
- Transfer pump

**Boil kettle**
- 6000W x 3 heating elements engaged by contactors
- Pump & Chiller with recirculation option for desinfication.

What I really need is a way to individual control the heating element I each kettle.

For the HLT when the temperature is nearing the correct temperature, I think only one element shout be used to keep the temperature constant. But if the temperature drops considerable it should engage 2 or 3 elements.

For the boil kettle something similar is needed, when nearing/reached 100 Celcius only one or 2 elements should be engaged depending on how vigorously boil is needed (and batch size).

Basically multi heating element control for each kettle depending on temperature, and pump re-circulation control (controlling valves & pump for each kettle).