craftbeerpi/craftbeerpi #121: Extra hardware implementation

Ridemywideglide, 2016-12-15:
Can you add the ability to switch several GPIO at the same time within a step?

As in, Temp reaches set point (would require previous issue with steps advancing fixed) then moves to next step which would kick on a circulating pump, and a heater (RIMS) at the same time?

Manuel83, 2016-12-15:
There is some special pump logic implemented. 
This switches the pump and the heater if the temperature is below the set point. This was provided by some CBPi user. 

https://github.com/Manuel83/craftbeerpi/blob/master/brewapp/base/automatic/pump.py

Ridemywideglide, 2016-12-16:
I can't get it to turn on the pin that it's assigned for the pump with pumplogic.

Ridemywideglide, 2016-12-17:
Anytime I choose pumplogic in kettle settings, it locks up when I turn on autopilot. Have to reboot the pi..

Ridemywideglide, 2016-12-17:
If it doesn't lockup, then it does nothing when on autopilot, not when you hit start.. 
I have step 1 set to heat to 110*, which works on PID or overshoot, but it won't turn on the heater when pumplogic is selected. I just does nothing.

Ridemywideglide, 2017-03-31:
Anyone actually get this to work??

SlabOch, 2017-04-12:
It would be really very nice to add this feature. It would best as a part of the step:
[image]

Twit3986, 2017-05-09:
This could be achieved by adding sub steps as per another enhancement request? You could create a new kettle called rims where the "heater" is the pump and target temp is >100.