craftbeerpi/craftbeerpi #122: i2c LCD Display

ollinator2000, 2016-12-17:
Hi ,
i'm wondering how to control a i2c 20x4 LCD Display out of the box with craftbeerpi 2.2.  At the moment i have a 2.1 setup running with the i2c solution from Hobbybrauer.de forum. 

i found the parameter in config.json and enabled it, but i guess thats not all .. ;-) 

may be i made a mistake. 

THX for your help 

Gretes from Hamburg
Oliver

Manuel83, 2016-12-20:
Hi,

can you provide me the code from Hobbybrauer.de forum. I will add it to version 2.2

JamFfm, 2016-12-21:
This would be perfect because 20x4 LCD modul will be my next Project. Will you add it to 2.2 or the master?

ollinator2000, 2016-12-22:
Sorry Pre Christmas insanity in the it business.. ;-) 
I'll provide the 3 Python Scripts for Manuel , so he can get rid of it.

BREWERNIE, 2016-12-23:
I would like to connect a lcd display too !

Greetings
Ernie

Manuel83, 2016-12-24:
Just received the code. I will have a look

Manuel83, 2016-12-24:
I have just created a new branch with first LCD test. 

The branch name is "lcd"

Unfortunately I have no LCD. 
Can someone test the code please?

git clone -b lcd http://github.com/manuel83/craftbeerpi

Futhermore you have to execute the following lines on you PI for installation. 

sudo apt-get install python-smbus i2c-tools
sudo modprobe i2c-dev
sudo modprobe i2c-bcm2708

Beginning from line 34 the displayed content can be cusomized

https://github.com/Manuel83/craftbeerpi/blob/lcd/brewapp/base/lcd/lcddisplay.py

Merry Christmas

blazem, 2016-12-27:
Yep, confirmed working.

Only thing is the LCD does not clear the screen before each screen update.

Perfect timing for me cheers Manuel.

Manuel83, 2016-12-28:
In Line 34 of lcddisplay.py is a python decorator and no invocation of clear().
Can you post your modified code please?
Maybe you missed () after "clear"?

BREWERNIE, 2016-12-28:
Hi Manuel,

i loaded an new library from here:
 git clone https://github.com/the-raspberry-pi-guy/lcd

Now it works !

Greetings 
Ernie

duzvik, 2016-12-30:
it supports 2 or 4 line display?

h44r3k, 2016-12-30:
It would also be a cool feature to use a rotary click encoder to enable scrolling through a simple menu and to do temp adjustments etc when using lcd

JamFfm, 2017-01-04:
Yep, confirmed working. Superb!
I had to enable the I2C in the Raspberry first.
Problem: contrast is very high so that I can nearly notice the letters. There is a screw to adjust the contrast but the lowest adjustment is still too high. LED Power is connected to 3.3 V of the Raspi. (v2)

JamFfm, 2017-01-04:
Looks like I need 5v to operate the LCD clearly. Don 't dare to connect it with 5 V. Ordered a level shifter board like this: 
https://www.amazon.de/gp/product/B00JQ0DITU/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1 
to protect the GPIO from 5v. Don't know if this makes sense but the save way. Another 3 weeks to wait.

JamFfm, 2017-01-08:
It works with 5v Connection. 
Very cool, to be able to watch the temp without a device like PC, iPad etc.
To be sure not to destroy the pi connect a Level Shifter.

ollinator2000, 2017-01-08:
mmhm i can only see the Header on my Display . ( craftbeerpi 2.2 & brewery name ) 
When i start the automatic Brewtask nothing more is displayed . 
Any tips of what i missed ? 

THX and Greetings
Olli

blazem, 2017-01-08:
Olli have you installed the drivers etc?

ollinator2000, 2017-01-09:
blazem: yes , and i upgraded the i2c_lib as mentioned before.  ( without that step the display shows nothing )
i can run the demo python script and everything works fine. it looks like the code never left lcddisplay.py 
[ if (app.brewapp_current_step == None): ] 

i guess i have to do some more investigation.. ;-)

ollinator2000, 2017-01-09:
unfortunately no success. my display shows only the first 2 lines. it stucks in current_step == none Mode. 8( 

there are also no error messages.. 

ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
INFO:socketio:emitting event "temp_udpdate" to all [/brew]
INFO:engineio:df351d6fac6b4dc49d4bc043f790242b: Sending packet MESSAGE data 2/brew,["temp_udpdate",{"2":22.12}]
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
INFO:socketio:emitting event "temp_udpdate" to all [/brew]
INFO:engineio:df351d6fac6b4dc49d4bc043f790242b: Sending packet MESSAGE data 2/brew,["temp_udpdate",{"2":22.12}]
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0
ERROR:brewapp:Exceptionlcdjob: 0

JamFfm, 2017-01-14:
In /home/pi/craftbeerpi/brewapp/base/lcd i fiddled around with the lcddisplay.py file. 

Now I implemented the wlan IP adress to be shown at the statup screen. 
When entering the steps (Button "Start" in steps) the display changes to show the daytime in addition to the stepdata. 
Furthermore there are no letters left when changing the mode. 
Got very low knowlage of Phyton or programming. So implementation is not perfect.

[lcddisplay.zip](https://github.com/Manuel83/craftbeerpi/files/706272/lcddisplay.zip)

Things to do: handle the german ß and the degree Symbol

And at least implementation LCD to the master branch

JamFfm, 2017-02-05:
update:
° is displayed now. 
Fixed issue when a String consists of more than 20 letters than it caused blinking of letters in the next line.
[lcddisplay.zip](https://github.com/Manuel83/craftbeerpi/files/753016/lcddisplay.zip)

JamFfm, 2017-02-16:
update:
Fixed issue when Temperature is only shown with one or no digits after the point (like xx.x  or  xx,  should be xxx.xx) the whole displayed temperature is no more in line.

[lcddisplay.zip](https://github.com/Manuel83/craftbeerpi/files/781388/lcddisplay.zip)

speedshifter2, 2017-03-09:
Hi I am having the same thing happen as ollinator2000, with only the first 2 lines showing on the display (craftbeerpi 2.2 & brewery name). When starting a Brew Task, the display does not change.
Is there something that i have missed? Any suggestions? 
Thanks.

ollinator2000, 2017-03-13:
Sorry , i m running the my 20x4 i2c Craftbeerpi Version still on Version 2.1 . I started constructing a new CBP simultanously, so i  decided not to analyze he Data anymore.  What i know , Version 2.1 is running perfectly with the i2c LCD .

blazem, 2017-03-13:
There are some drivers you will need to install to get the LCD working
which is on the documentation

arcidodo, 2017-04-16:
Is it possible to get the countdown timer on the LCD?

oleost, 2017-10-06:
What is current status with display in craftbeerpi 3?

I've got one of these https://m.banggood.com/0_96-Inch-4Pin-Blue-Yellow-IIC-I2C-OLED-Display-Module-For-Arduino-p-969144.html laying around, it would be handy to be handle to show Target Temp, Actual Temp and time left.

ollinator2000, 2017-11-01:
LCD Display works great with LCD Add-on on CBP3

closed 2017-11-01 by ollinator2000