craftbeerpi/craftbeerpi #123: Boil logic

oleness, 2016-12-20:
Hello

I'm currently using an ArdBir controller, and my plan is to migrate to CraftBeerPi eventually.

The ArdBir controller goes to PWM at a set boil-temperature instead of using PID during boiling. This makes it easy to control the rolling boil by setting the heating element on-cycle by adjusting the PWM %

Also I would like "sub-events" during boiling like hop-additions and reminders with buzzer alarms ;)

Manuel83, 2016-12-20:
Valid point. Currently its not implemented I like to add a more flexible step / sub step concept to CBPi.

Xmas holiday is coming. Means time to work on CBPi :-)

Ph03n1X1, 2017-01-12:
There are now 2 pull requests with slightly versions of implementations for PWM. Implement at your own risk. Mine has been tested on hardware, but not actually doing a boil.