craftbeerpi/craftbeerpi #126: Unable to access Craftbeerpi v2.2 via ios/macos

mactore88, 2016-12-31:
I am having issues accessing Craftbeerpi v2.2 from my ios devices and my macbook. No problem acessing it via my android smartphone or PC. I have just done a clean install of the new software, but still no luck in accessing it. Does anyone have a solution for this?

duzvik, 2016-12-31:
nope, today i've made a clean install and everything is OK. Except temperature  graphs :(
Check that mac is connected to same wi-fi network.

DrLaban, 2017-01-11:
Did you figure out the graphs @duzvik ? Delete the tempsensor logfiles - then the graphs should work again.

duzvik, 2017-01-11:
@DrLaban Thanks. This helps. 
Same here:
https://github.com/Manuel83/craftbeerpi/issues/113