craftbeerpi/craftbeerpi #127: Fermentation steps do not restart after power loss.

JustBos, 2017-01-01:
I have noticed that if the Pi loses power then the fermentation process does not restart, you must log back in manually restart. 

Is this by design or it is meant to restart and continue from where it left off?

Cheers

Manuel83, 2017-01-01:
I will check this

JustBos, 2017-01-02:
Awesome, thanks.

aedocw, 2017-01-06:
I would also love "resume previous state" after power loss/reboot.

nambuco, 2017-02-15:
I have the same issue. Will version 3.0 solve this issue?

nambuco, 2017-02-19:
I made some changes in the fermenter.py and the problem is solved. Now, after reset, if there is an active fermentation step running, automatic mode is activated.

I do not know how to share the changes I made.

nambuco, 2017-02-19:
Basically I changed following functions:

```
@brewinit()
def load():`
    for s in Fermenter.query.all():
        app.cbp['FERMENTERS'][s.id] = to_dict(s)
    for s in FermenterStep.query.filter_by(state='A').all():
        app.cbp['CURRENT_TASK'][s.fermenter_id] = to_dict(s)
        app.cbp['CURRENT_TASK'][s.fermenter_id]["autoreset_required"] = 'Y'
        if s.state == 'A' and s.timer_start is not None:
            app.cbp['CURRENT_TASK'][s.fermenter_id]["endunix"] = int((s.timer_start - datetime.datetime(1970, 1, 1)).total_seconds())

    app.logger.info("CURRENT_TASK")
    app.logger.info(app.cbp['CURRENT_TASK'])
```
```
@brewjob(key="fermenter", interval=60)
def fermenterjob():
    for id in app.cbp['FERMENTERS']:
        fermenter = app.cbp['FERMENTERS'][id]
        temp = app.brewapp_thermometer_last[fermenter["sensorid"]]
        timestamp = int((datetime.datetime.utcnow() - datetime.datetime(1970, 1, 1)).total_seconds()) * 1000
        writeTempToFile("F_" + str(fermenter["id"]), timestamp, temp, fermenter["target_temp"])
    for i in app.cbp['CURRENT_TASK']:    
        if app.cbp['CURRENT_TASK'][i]["autoreset_required"] == 'Y':
            app.logger.info("autoreset_required")
            fermenter_automatic(str(i))
            app.cbp['CURRENT_TASK'][i]["autoreset_required"] = 'N'
```