craftbeerpi/craftbeerpi #131: PiFace issue on 2.2

bjornarfonn, 2017-01-04:
After upgrading from developer to final 2.2 my PiFace is not working. 
Ran the server manually and when activating the switch, it prints;

> File "/home/pi/craftbeerpi/brewapp/base/devices/piface.py", line 36, in switchON
>     gpio = self.translateDeviceName(device)
>   File "/home/pi/craftbeerpi/brewapp/base/devices/piface.py", line 32, in translateDeviceName
>     pin = int(name[6:])
> ValueError: invalid literal for int() with base 10: ''

Just after starting the application, it prints the following and keeps repeating;

> [2017-01-04 23:22:50,819] ERROR in __init__: Exceptionkettlejob: int() argument must be a string or a number, not 'NoneType'

bjornarfonn, 2017-01-06:
So, just re-installed rasbian and cbp, still no luck. The software switch doesnt light up green either when engaged. PiFace testing from terminal shows that its working correctly.

BREWERNIE, 2017-02-17:
My Piface also does not work at all on the version 2.2.

bjornarfonn, 2017-04-29:
Thought i would get some action on this issue again. I had initially abandoned the PiFace interface for a while, but as i was building yet another controller, i tried it again with the master branch, and still doesnt work. I cleared the install and installed dev 2.1 and it works again. So, something (above my knowledge) happened with the new version 2. Its something that i guess is a reference flaw in the coding, would be happy to try out some troubleshooting to get this resolved.

olhaug, 2017-05-11:
I also have problems. It seems like my SSR doesn't work at 3.3v via the GPIO, but it seems like i need 5v at that's not possible from GPIO so i need to use the Piface to switch the SSR.