craftbeerpi/craftbeerpi #135: Hysteresis mode doesnt work

fuzzewuzze, 2017-01-09:
I tried my first CBP brew this weekend with great success, until it came time to boil.

My panel is my old SBE panel, so lets get wiring issues out of the way as thats not it(and PID works)

Steps i followed

Turn BK Switch on Panel ON
Edit Boil Kettle, set it to Hysteresis mode
Set ON = 2, SET OFF = 2
Apply
Push car button( Still dont understand what this means but it seems to be auto control?), my element heater turns on, but never turns off.

I was not sure if this was in Milliseconds, so i also tried 2000 on and 2000 off, same thing.  

This is a pretty big issue IMO, as i was stuck boiling 100% with a 5500W element which makes a pretty violent boil.

Ph03n1X1, 2017-01-10:
I assume the car button is supposed to be a universal symbol for "auto", meaning automatic rather than manual control of the heater. Also, I would be interested to see if this issue is resolved. Today, I just bought a separate controller to work around the issue of not being able to control the heating element for boil to give a soft boil.

fuzzewuzze, 2017-01-10:
Out of all them i'd think it would be the easiest so i hope your right, its literally on/off every X seconds.

Ph03n1X1, 2017-01-11:
I'm not sure what hysteresis mode is intended to do, but I read through the code and all it does is turn the heater on if the temp drops the number of degrees below the set temp minus the "on" value or turns the heater off if the temp rises the number of degrees above the set temp plus the "off" value.

schrod has a pull request for a PWM add-in that should work for boil control and I have developed another that I will be submitting soon. You can manually add in one of these to your code until/unless they are pulled into the main branch (at your own risk, of course).

fuzzewuzze, 2017-01-11:
I wont be brewing for a few weeks probably, so hopefully its all figured out by then.

I assumed the Hysteresis worked like that in SBE, which i guess is more PWM.

pr0woKator, 2017-02-01:
Did you solve your problem?

Ph03n1X1, 2017-02-01:
There wasn't really a problem here. The issue reported was a misunderstanding about how a certain logic mode worked. A mode that works as fuzzewuzze requested (PWM) has been created and is in pull request status.