craftbeerpi/craftbeerpi #159: LED when device is ready

olemde, 2017-03-11:
Hi 
Does antöne know how i can implement an LED which should be ON when the Craftbeerpi is "ready" - after the boot Process is finished?

Thx

Ole

Manuel83, 2017-03-11:
Hi,

you could create a custom init method. 
All methods decorated with @brewinit() will be executed once after startup.
You can set a order number like 1000. This defines the order in which all init methods are executed. Just take a very high number :-)

If you search in the CBPi repository for @brewinit you will find some examples.

```
@brewinit(1000)
def startupcomplete:
     # YOUR LED CODE GOES HERE
```

olemde, 2017-03-14:
Hi Manuel
thank you for support. sounds very easy - anyhow, i am not able to program a GPIO-pin in python in that manner. To implement the code in a py-file i am able but not in creating it..
Does someone wants to help me? it would be great. 
Thanks for Manuel anyway for that great brewing tool

Ole

archonic, 2017-04-03:
Try this from your home dir, which is the default when you start terminal or SSH in as the pi user
`sudo nano craftbeerpi/brewapp/base/automatic/automaticlogic.py`

Then paste this at the very bottom:

    @brewinit(1000)
    def startupcomplete:
      GPIO.setup(12, GPIO.OUT)
      GPIO.output(12,True)

Change 12 to whatever pin you have your ready LED wired to.

@Manuel83 Is there a particular folder or file for custom code like this? I didn't see anything in particular in the .gitignore. Not a bad idea to create brewapp/base/custom/* or something like that to hook into your decorators and not get overwritten when updating.