craftbeerpi/craftbeerpi #163: PIO on every 6th second

hjarupBrewDK, 2017-03-17:
Using CBP2.2, PIDlogic , with a regular relay.

I measure/observe short on burst 10mS duration in the relay off periods.
This happens every 6th second.

This shortens the relay life, and seems unnecessary. Unless it is used for creating a pause to read the TEMP sensors? (EMI performance)

(same behavior on etc. PIDarduinoLogic)
Is this a known issue ?

hirschmann, 2017-04-01:
It's more or less a bug which can easily be fixed.
I've created a temporary fix for PIDAruinoLogic, checkout https://github.com/hirschmann/craftbeerpi/commit/986f39797bc68ed176d2199127fcd60b80683d2d

In case you're not familiar with git, you can also replace _pid_aruino.py_ as a whole with the fixed version: https://github.com/hirschmann/craftbeerpi/blob/tmp-fix-issue-163/brewapp/base/automatic/pid_arduino.py

Let me know if this solves your problem :)

@Manuel83
I'll create a proper pull request which fixes PIDArduinoLogic and PIDAutotuneLogic soon.
Let me know if I should fix PIDLogic also.

hjarupBrewDK, 2017-04-01:
I am not to strong in software, more in hardware. so i replaced the pid_aruino.py as a whole. It fixes the issue on PIDAruinoLogic.