craftbeerpi/craftbeerpi #175: PIDAutotune do not write data in pidparams.txt

cocothenut, 2017-05-01:
Hi

using the last branch fix-ssr-flickering

1.

i had run autotune now several times , but there is no output ( data ) write in the pidparams.txt

/home/pi/craftbeerpi is on 777 

using the branch without the "fix-ssr-flickering" pid-data was write in the pidparams.txt 

2.

what is :

output step % -- how to use / adjust this ?

lookback seconds -- how to use / adjust this ?

max.output % - is limiting your heater's maximum power output (i find it)

3.

what are the different "rules" -- how to use ?

pessen-integral
ziegler-nichols
brewing
and so on

musst i change the rule somewhere in the code ?
and which (rule) pidparameters i have to put in the PIDArduinoLogic ?

Thanks for any Help

webgestalt, 2017-05-01:
I have the same problem. pidparams.txt isn't written....... After 2h I killed the process and CBPi was in Automatik Mode 😔

hirschmann, 2017-05-02:
@cocothenut
1. I didn't change the autotune-algorithm itself in the *fix-ssr-flickering*-branch.
It should work as before.
2.
    - **output step %**: defines the the output of the autotune-algorithm when stepping up/down, e.g. *output step* = 100; step up (=heating) output = 100; step down (= cooling) output = -100. This setting should stay at 100%
    - **lookback seconds**: determines how far the algorithm will look back when trying to find local (temperature) extrema (minima/maxima). If the algorithm recognizes even short peaks as extrema, you should increase this value. If it doesn't recognize actual extrema, you should decrease it. Usually the default of 30 seconds work fine.
    - **max. output %**: limits the maximum power output. This is useful if your heater is overpowered and would heat up the kettle way too fast. If you don't want to limit your heater, leave this at the default value of 100%
3. The PID-tuning rules basically are different ways of calculating the PID-parameters, which changes the behavior of the PID-controller (e.g. ziegler-nichols produces quite a lot of overshoot - not that good for brewing). See [Ziegler–Nichols method](https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method).
In most cases, *brewing* is the best rule for brewing, obviously ;)
You don't have to change anything in code. Just copy the *brewing*-parameters into the PIDArduinoLogic's settings.

The autotune-process can fail. In this case, the automatic mode will be disabled, but no *pidparams.txt* file will be created.
If autotune runs forever, the problem is usually either the heater being not powerful enough to reach the setpoint or the temperature not dropping below the setpoint fast enough for whatever reason (isolation, ambient temp. etc.).

If auto-tuning doesn't work for you, please do the following:
- stop craftbeerpi
- delete all logfiles in `craftbeerpi/log/`
- start craftbeerpi
- select PIDAutotuneLogic
- enable automatic mode and wait

You can read the autotune log by reading the (newly created) logfile via:  
`tail -f craftbeerpi/log/name_of_the_logfile | grep PIDAutotune`  
If you don't know how to interpret the log, please upload the whole logfile here (as a file, please don't paste the content).

You can also checkout my [PID/Autotune simulation](https://github.com/hirschmann/pid-autotune), which yields quite good PID-Parameters in a matter of seconds :)

cocothenut, 2017-05-02:
Hi

thanks for your reply and explanations

the autotune-process itself runs completly (take some time) , the pidparams.txt was created , but no PID-parameters are writen in the pidparams.txt (pidparams.txt is empty)

with the "old" files (pid.py , pid_arduino.py , pid_autotune.py) the pidparams.txt was filled with the PID-parameters

with the "new" (fix ssr flickering) files the pidparams.txt stays empty

my workaround for testing is now :

running autotune-processes with "old" files , saveing pidparams.txt
copy "new" files in brewapp/base/automatic , put PID-parameters into PIDArduinoLogic
running working tests

so far it is working well , now doing some fine-tuning

hirschmann, 2017-05-02:
Should be fixed now, give it a try: https://github.com/Manuel83/craftbeerpi/pull/171/commits/51135a137cd7d6ee928babd47ba5a91c3271c80a

cocothenut, 2017-05-05:
Hi

after some tests 

with the "new" (fix ssr flickering) files and the fixed pid_autotune.py ( 51135a1 )

pidparams.txt was filled with the PID-parameters as desired

the autotune process takes approximately double of time as with the "old" files ?

so i had to put double amount of beer bottles in my freezer for waiting-time during the autotune process ;-)