craftbeerpi/craftbeerpi #178: PIDAutotune writing date,time,settings

cocothenut, 2017-05-10:
hi

is it possible to ad date , time , settings 

to the pidparams.txt ?

would be more efficient to archive by running several autotune tests

like

10.05.2017  15:00
target-temp : 70
output step % : 100
max.output % : 75
lookback seconds : 30

rule: ziegler-nichols
Kp: 7.66141138481997
Ki: 0.4086086071903984
Kd: 35.91286586634361

rule: tyreus-luyben
Kp: 5.920181524633613
Ki: 0.07104217829560336
Kd: 35.239175741866745

and so on

Thanks a lot

hirschmann, 2017-05-10:
Good idea! I've updated PR #171 

If autotuning succeeds, the resulting parameters for all rules will be appended to *pidparams.txt * with the following header:

```
created at: 2017-05-10 19:53:06.419878
target temperature: 60
output step %: 100.0
max. output %: 75.0
lookback seconds: 30.0
```

cocothenut, 2017-05-12:
hi

sorry no data in pidparams.txt writen

pidparams.txt was created but not filled with data

hirschmann, 2017-05-12:
I've modified the PR, please try again. PIDAutotuneLogic will always write a result to pidparams.txt no matter whether tuning failed or succeeded.

cocothenut, 2017-05-13:
Hi

now it works , end-user-friendly   :-)

pidparams.txt was created with the data inside 

tuning started at: 2017-05-13 11:40:16.718926
tuning finished at: 2017-05-13 13:09:04.621248
target temperature: 70
output step %: 100.0
max. output %: 75.0
lookback seconds: 30.0
result: tuning succeeded

rule: pessen-integral
P: 1.4949975085
I: 0.065090358864
D: 12.9087037846

rule: ziegler-nichols
P: 1.23117441876
I: 0.0428830599575
D: 8.83676707606

rule: brewing
P: 16.7439720951
I: 0.0874814423132
D: 50.6021188355

and so on

Thanks