craftbeerpi/craftbeerpi #184: Craftbeerpi Wik not start

geirij88, 2017-06-01:
My raspberry pi 3 had a crash last week, and I am now trying to get CraftBeerPi up and running again but it will not start.  I have reinstalled raspian os and done the installation as it says on the video on CraftBeerPi homepage. CraftBeerPi server is running, but I am not able to start the webpage on port :5000.

brodiep, 2017-06-08:
I am new here and have installed Craftbeerpi on a clean install of Rasppi. I think the server is running but this is what I get I ask for the status:

> Debian

 GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jun  8 11:52:30 2017
pi@raspberrypi:~ $ sudo /etc/init.d/craftbeerpiboot status
● craftbeerpiboot.service - LSB: Put a short description of the service here
   Loaded: loaded (/etc/init.d/craftbeerpiboot)
   Active: active (exited) since Thu 2017-06-08 11:52:18 AEST; 16min ago
  Process: 519 ExecStart=/etc/init.d/craftbeerpiboot start (code=exited, status=0/SUCCESS)

Jun 08 11:52:18 raspberrypi systemd[1]: Started LSB: Put a short description....
Jun 08 11:52:18 raspberrypi craftbeerpiboot[519]: Starting system CraftBeerPI...
Hint: Some lines were ellipsized, use -l to show in full.
pi@raspberrypi:~ $

brodiep, 2017-06-08:
Just a little more:
"Last login: Thu Jun  8 12:06:28 on ttys000
Pauls-Mac-mini-4:~ paulbrodie$ ssh pi@192.168.15.18
pi@192.168.15.18's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jun  8 12:08:33 2017 from 192.168.15.5
pi@raspberrypi:~ $ cd craftbeerpi
pi@raspberrypi:~/craftbeerpi $ sudo python runserver.py
Traceback (most recent call last):
  File "runserver.py", line 3, in <module>
    from flask import Flask
  File "/usr/local/lib/python2.7/dist-packages/flask/__init__.py", line 19, in <module>
    from jinja2 import Markup, escape
ImportError: No module named jinja2
pi@raspberrypi:~/craftbeerpi $ "

brodiep, 2017-06-09:
Manuel, I went back to the drawing board with a complete reinstall of the PI and your app, it is all working and I must say well done, a great job. I am looking at coupling your software to a Keg King RoboBrew, I will keep you posted.