craftbeerpi/craftbeerpi #185: Step timer countdown problem

kranja, 2017-06-28:
I'm reporting a problem with the step countdown timer which is fixed on 00:00:00 value when the step starts and can't be started manually.
Some people reported this issue also with v 3.0 on the fb group.

kranja, 2017-07-10:
Not sure if helps, but on the last brew I noticed that the timer is working and is visible if I start the step on a rpi touchscreen. At the same time on a wlan connected tablet there was no timer counting down. I tried to start the next step from the tablet and again nothing happened.

tstrooba, 2017-07-29:
We are also having this issue.  Sometimes it happens sometimes it doesn't.  It started when refreshing the page but today this did not work.  We're not on version 3 yet.

kbaggen, 2018-03-11:
Hi there, I also got this issue!
When I go till "NEXT STEP" the time is showen suddeenly, hence, I can see how long I was at that first step afterwards, but there is no running count down!
BTW....thanks for thi amazing software!

bitbot-xx, 2018-06-30:
Yup it's broken. Also start/end times don't have dates. And the logs are enormous so not helpful.