craftbeerpi/craftbeerpi #187: Version 2.2, trouble with temp sensors, WHY?

Nauticno, 2017-07-18:
Hello!

My first brew with CraftBeerPI stopped due to temp sensore problem. I have tested the sensors. I have deleted craftbeerpi.db with no luck. Tried with reinstall with no luck. I have tried Git pull to check for updates.

Here is status; 
pi@raspberrypi:~/craftbeerpi $ sudo /etc/init.d/craftbeerpiboot status
● craftbeerpiboot.service - LSB: Put a short description of the service here
   Loaded: loaded (/etc/init.d/craftbeerpiboot)
   Active: active (running) since Tue 2017-07-18 15:21:10 CEST; 8min ago
  Process: 865 ExecStop=/etc/init.d/craftbeerpiboot stop (code=exited, status=0/SUCCESS)
  Process: 904 ExecStart=/etc/init.d/craftbeerpiboot start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/craftbeerpiboot.service
           └─909 python /home/pi/craftbeerpi/runserver.py

Jul 18 15:21:10 raspberrypi craftbeerpiboot[904]: Starting system CraftBeerPI daemon:.
Jul 18 15:21:10 raspberrypi systemd[1]: Started LSB: Put a short description of the service here.

I find:
pi@raspberrypi:~ $ ll /sys/bus/w1/devices
total 0
lrwxrwxrwx 1 root root 0 Jul 18 15:22 24-4ff3f0258150 -> ../../../devices/w1_bus_master1/24-4ff3f0258150
lrwxrwxrwx 1 root root 0 Jul 18 15:22 93-8c8bf5a5d154 -> ../../../devices/w1_bus_master1/93-8c8bf5a5d154
lrwxrwxrwx 1 root root 0 Jul 18 15:22 f8-22f73535e9ab -> ../../../devices/w1_bus_master1/f8-22f73535e9ab
lrwxrwxrwx 1 root root 0 Jul 18 15:16 w1_bus_master1 -> ../../../devices/w1_bus_master1

but RaspBerryPI just finds 2:
28-041661e7fcff
28-031661f8cdff

Need help!!!

Nauticno, 2017-07-18:
I startet With a clean card and coplete reinstall! Works now!

archonic, 2017-11-14:
@Nauticno Great. Could you close the issue please? Thanks!