craftbeerpi/craftbeerpi #194: CBP won't run.  runserver.pi not running?

storunner13, 2017-11-02:
Hello--I've installed then uninstalled and reinstalled 3 times with this same issue.  How can I correct?  I've tried looking online for solutions, but none seem to fit my issue.  I select "autostart = y" and then reboot, but there is no autostart.

Then in the directory I try `sudo /etc/init.d/craftbeerpiboot start`
Which results in `[ ok ] Starting system CraftBeerPI daemon:.`

I then check the status `sudo /etc/init.d/craftbeerpiboot status`
resulting in `[FAIL] /home/pi/craftbeerpi/runserver.py is not running ... failed!`

What is going wrong? How can I correct this?

hjarupBrewDK, 2017-12-13:
Hi , I recommend you read this section - it have been updated today: https://github.com/Manuel83/craftbeerpi/wiki/First-Installation

Br Carsten

pauloviniciusfernandes, 2020-02-22:
Hello. I have a problem too. My craftbeerpi3 won't run. In the installation show the red words  " gitdb2 riquires Python'>=3.4' but the running Python is 2.7.16 " 
I try install a lot debians version, I try make update and don't make up date and It won't run.