craftbeerpi/craftbeerpi #197: Connection refused|

Veigrillen, 2017-11-25:
Hi guys.
Im a noob to Raspberry, and hope to get some help here.

I have installed raspbian and craftbeerpi, but when i try to connect to localhost:5000 or <IP>:5000 i get an message saying refused to connect.

Dont know if this is an issue iwe got with craftbeer or my raspberry or what.

Anyone got the time to help me?

thanks in advance

hjarupBrewDK, 2017-12-12:
Hi All, I installed fresh today - with the same result. The installation is succesfull - however the webserver is not started. And does not start via manual command.

I guess next step is to find the installation LOG file, and see if any part did not go well.

Br Carsten

hjarupBrewDK, 2017-12-12:
Some more results from command line testing:

pi@raspberrypi:~ $ sudo /etc/init.d/craftbeerpiboot start
[ ok ] Starting craftbeerpiboot (via systemctl): craftbeerpiboot.service.
pi@raspberrypi:~ $ sudo /etc/init.d/craftbeerpiboot status
● craftbeerpiboot.service - LSB: Put a short description of the service here
   Loaded: loaded (/etc/init.d/craftbeerpiboot; generated; vendor preset: enabled)
   Active: active (exited) since Tue 2017-12-12 06:17:27 UTC; 13h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 336 ExecStart=/etc/init.d/craftbeerpiboot start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/craftbeerpiboot.service

Dec 12 06:17:25 raspberrypi systemd[1]: Starting LSB: Put a short description of the service here...
Dec 12 06:17:27 raspberrypi craftbeerpiboot[336]: Starting system CraftBeerPI daemon:.
Dec 12 06:17:27 raspberrypi systemd[1]: Started LSB: Put a short description of the service here.
Hint: Some lines were ellipsized, use -l to show in full.
pi@raspberrypi:~ $ netstat -plnt
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -
tcp6       0      0 :::22                   :::*                    LISTEN      -

HooTsBrooklyn, 2017-12-12:
Is there any reason why you're not using version 3?
https://github.com/Manuel83/craftbeerpi3

hjarupBrewDK, 2017-12-12:
I grabbed the log from the install, I hope someone can use it to find out the issue. 

[log_info.txt](https://github.com/Manuel83/craftbeerpi/files/1553472/log_info.txt)

HooTsBrooklyn, 2017-12-12:
Try "sudo pip install -r requirements.txt"
Also,  "sudo apt-get update && sudo apt-get upgrade" then run the install again

Looks like pip / Python isn't fully installed with all requirements.

hjarupBrewDK, 2017-12-12:
Tried the two commands, see log file.
[log1.txt](https://github.com/Manuel83/craftbeerpi/files/1553593/log1.txt)

HooTsBrooklyn, 2017-12-12:
Sadly, this is beyond my abilities.  I've only used CBPi 3.  There is not reason to install 2 if you're upgrading to 3.  You might be able to get faster results from the CBPi Facebook Group

hjarupBrewDK, 2017-12-13:
Got it, thanks. I will ask on FB.

Veigrillen, 2017-12-13:
I didnt use v3 because im a newbie 🤣
I just made an reinstall and it all worked. 😃

hjarupBrewDK, 2017-12-13:
If downloading the Jessie version of Raspbian it will works.

I tested it selecting NO to the first question regarding updating in the CBP installer.

http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/2017-07-05-raspbian-jessie.zip

Thanks to all in the FB forum for helping out. I plan to update the wiki with this infromation for CBP2.2

hjarupBrewDK, 2017-12-16:
wiki updated with this newfound information https://github.com/Manuel83/craftbeerpi/wiki/First-Installation

blucmal, 2018-01-29:
So for v3 do we still need the jessie version? I installed rasbian via noobs the craftbeerpiv3 and am getting connection refused via lan or localhost/127.0.0.1....

HooTsBrooklyn, 2018-01-29:
If you're having troubles getting CBPi 3.0 up and running, try following the steps outlined here
https://cbpi.mricesolutions.com/viewtopic.php?f=3&p=18

blucmal, 2018-01-30:
I followed your advice above to reinstall? Python? And it fixed it. Thanks for the link will check it out.

HooTsBrooklyn, 2018-01-30:
You prolly won't need to reinstall, but just install the python PIP modules.