craftbeerpi/craftbeerpi #228: Unable to log onto cbpi after install.

ianherbertbeer, 2020-01-08:
Hi
I’m new and I’m trying to set cbpi for the first time on a pi4. Ive been thought the install twice and I’m unable to log onto cbpi at all even after a reboot. 
Any ideas.

sergey8922, 2020-01-14:
The same problem.  
After the update does not start.  
No rights.

bullitt186, 2021-02-18:
I had on my Raspberry Pi Zero W with Raspberry OS Buster the same issue.
The root cause was, that apparently craftbeerpi3 relies on Python 2.7 which is deprectated.
The trick for me was to install python2 via "sudo apt install python2" and make it the default python version as described here: https://raspberry-valley.azurewebsites.net/Python-Default-Version/

Running the installation again after this change did the trick for me.